OpenTabletDriver/OpenTabletDriver #402: Wacom FT-0405-U

OmegaRogue, 2020-10-26:
## Description

It would be nice to have support for the Wacom Volito (FT-0405-U), as the latest drivers from wacom supporting it is version 5.2.5-5a

ssccoonniicc, 2020-10-26:
Open OTD, then OTD -> Help -> Export Diagnostics

Upload the generated file here

OmegaRogue, 2020-10-27:
github doesnt support uploading json, so here it is:
https://gist.github.com/OmegaRogue/3baa7f5d61fe440e55179bdb5b01debd

X9VoiD, 2020-10-29:
Use this [dev build](https://github.com/InfinityGhost/OpenTabletDriver/actions/runs/330614943) (sign in to GitHub to access download link), and then extract the zip below to `Configurations\Wacom`.

[FT-0405-U.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5460519/FT-0405-U.zip)

Please do note that this is just a preliminary config to properly detect the tablet and the dimensions in it are likely to be wrong.

OmegaRogue, 2020-10-30:
It detects the tablet, but has an invalid area size, so it seems to be working, I configured some approximate values for the dimensions and it seems to be mostly working, except that the cursor is flickering between the position it's supposed to be in and the top left corner of the screen area

X9VoiD, 2020-10-30:
Close all instances of OpenTabletDriver.UX and OpenTabletDriver.Daemon, then try again.

Make sure you have no other tablet drivers running in the background. When you get OTD up and running normally, open OTD -> Tablet -> Tablet Debugger, then hover the pen to the bottom rightmost part of the physical tablet area. Try to get the highest possible value for `Position` then report it here, also try to get the highest possible value for `Pressure`.

X9VoiD, 2020-10-30:
Also take physical measurements of your wacom tablet in `mm unit`

OmegaRogue, 2020-10-30:
Width: 128mm 5104px
Height: 98mm 3712px
pressure: 512

OmegaRogue, 2020-10-30:
i fixed the flickering with this config, there are multiple possible report ids depending on the status of the pen and what buttons are pressed https://gist.github.com/OmegaRogue/368d76b830e4651f2ffd4eef05d748e2

X9VoiD, 2020-10-30:
Are you sure that <5104, 3712> is the highest possible value for Position? Because that seems too small for its size.

For 2040LPI, it should be around <12800, 9800> if the physical measurement is accurate
For 1020LPI, it should be around <6400, 4900>

And for pressure, make sure that it really is 512 instead of 511
0 is also counted up to 511 to have 512 levels of pressure

X9VoiD, 2020-10-30:
And also
```
      "ActiveReportID": {
        "Start": 72,
        "StartInclusive": true,
        "End": null,
        "EndInclusive": false
      },
```
is enough to cover all of the other `DigitizerIdentifiers` as ActiveReportID specifies a range. The other DigitizerIdentifiers won't be loaded at all since they have the same identifiers as the first one.

InfinityGhost, 2020-10-30:
Reopening to link this issue to #410

closed 2020-10-30 by InfinityGhost