OpenTabletDriver/OpenTabletDriver #403: Gaomon S620 Driver issues

Manson-22, 2020-10-27:
drivers dont work with Gaomon S620. 
[image]
[image]

ssccoonniicc, 2020-10-27:
Do the following steps:
Device Manager -> View -> Devices by Connection
Find for any of these devices inside usb controllers:
  -  Gaomon HID
  -  Huion HID
  -  Graphics Tablet Device
  -  HID Tablet

Uninstall them (and check delete driver files) then replug tablet

Manson-22, 2020-10-27:
thanks, im not really the best with drivers, as I only know some basic info on the original hawku drivers, and I just got a new tablet.f

Manson-22, 2020-10-27:
never mind. it says invalid area size.

ssccoonniicc, 2020-10-27:
can you share a screenshot of the driver

InfinityGhost, 2020-10-27:
If you get that, reset to default settings, or change width and height to something bigger than 0 and set the position value to be a valid number

InfinityGhost, 2020-10-30:
Any update on this?

InfinityGhost, 2020-11-11:
Assuming this is resolved because of no response in over 7 days.

closed 2020-11-11 by InfinityGhost