OpenTabletDriver/OpenTabletDriver #404: Support for Genius i608x tablet

Kijekin, 2020-10-28:
## Description
I'd like to see a support for Genius i608x tablet here, or a sort of template for me to do it myself.

X9VoiD, 2020-10-29:
Uninstall all tablet drivers associated with the device, restart, then OTD -> Help -> Export Diagnostics, upload the file here.

X9VoiD, 2020-11-01:
Any news on this?

Kijekin, 2020-11-04:
Was away from home, gimme sec

Kijekin, 2020-11-04:
Is it correct?

[diagnostics.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5487276/diagnostics.zip)

X9VoiD, 2020-11-04:
[MousePen i608X.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5488659/MousePen.i608X.zip)

Extract zip contents to `Configurations\Genius` then restart OTD

X9VoiD, 2020-11-04:
Please do note that the tablet dimensions and max sensor values might not be accurate and are only there as dummy values to let OTD detect the tablet. Those values will be filled in later when OTD Tablet Debugger have detected it and starts receiving reports.

Kijekin, 2020-11-04:
It doesn's seem to detect it. Shall i do any other steps in that case?

X9VoiD, 2020-11-04:
Make sure that you've restarted OTD and no other configs reference the same i608X.

Then OTD -> Help -> Export Diagnostics

Kijekin, 2020-11-04:
Done:

[diagnostics2.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5489500/diagnostics2.zip)

X9VoiD, 2020-11-04:
Delete your current OTD, then download and extract [latest dev build](https://github.com/InfinityGhost/OpenTabletDriver/actions/runs/342954837) to any folder except admin requiring folders. Add the configuration back then try again.

Kijekin, 2020-11-04:
Ok, it detects the tablet now

X9VoiD, 2020-11-04:
Check Tablet Debugger and see if OTD actually opened the correct `Endpoint` by seeing if `<Position>` is correct

X9VoiD, 2020-11-04:
Most tablets have upper left as 0,0 and bottom right as the sensor's maximum value, around `10000 px/mm` on 2540 LPI

Kijekin, 2020-11-04:
Tablet debugger shows only four blank boxes

Kijekin, 2020-11-04:
Can it be connected to fact that system reads tablet as multiple instances instead of one?

X9VoiD, 2020-11-04:
Not the correct wording but yes, Windows split up the tablet into different HID usages, and all of them is similar from OTD's perspective, so we have no choice but to iterate over all of those until we succeeded opening one. And the problem with it is exactly this. It did not open the correct `endpoint`

Kijekin, 2020-11-04:
So how to iterate them? OTD seems like trying to open the same endpoint over and over again

X9VoiD, 2020-11-04:
OTD is opening the same one since it's the first one that opened successfully.

Kijekin, 2020-11-04:
It's the first one on the list that is matching its identifier

X9VoiD, 2020-11-04:
You'll have to modify OTD itself to manually change the endpoint

Kijekin, 2020-11-04:
in which section of the code shall i look for endpoint handling?

X9VoiD, 2020-11-04:
https://github.com/InfinityGhost/OpenTabletDriver/blob/master/OpenTabletDriver/Driver.cs#L76

Kijekin, 2020-11-04:
Thanks a lot, I'll dive into it tomorrow.

X9VoiD, 2020-11-05:
@Kijekin progress is made for this issue in #425, feel free to try it out.

An example Attribute to match HID Usage would be:

```
  "Attributes": {
    "WinUsage": "08"
  }
```

to match `\\\\?\\hid#vid_0458&pid_501a&col08#6&3e23d5&5&0007#{4d1e55b2-f16f-11cf-88cb-001111000030}` (col**08**)

X9VoiD, 2020-11-13:
Any news on this?

KoleckOLP, 2020-12-03:
Hello I'd like to ask if this would also make the **Genius WizardPen 5x4** work too or should I make a config request once I get home and get more info about the tablet (such as the diagnostic)

InfinityGhost, 2020-12-04:
> Hello I'd like to ask if this would also make the **Genius WizardPen 5x4** work too or should I make a config request once I get home and get more info about the tablet (such as the diagnostic)

Make its own configuration request, its easier to track when they're separate.

vladfrangu, 2020-12-08:
Hey, sorry to chime in on this, I have the exact same tablet, what should I try out to use this driver? 👀

InfinityGhost, 2020-12-09:
> Hey, sorry to chime in on this, I have the exact same tablet, what should I try out to use this driver? eyes
> 
> If it helps, my diag JSON is: https://gist.github.com/vladfrangu/274d1cfd762f0e979512ca4981c85c8a

Pretty much repeat the same steps @X9VoiD said to the issue creator, and see if stuff works.

vladfrangu, 2020-12-09:
I followed the first step of adding the config and the tablet gets picked up, but as seen in the logs, it doesn't connect. Now, I assume following this comment is what I need to do

> @Kijekin progress is made for this issue in #425, feel free to try it out.
> 
> An example Attribute to match HID Usage would be:
> 
> ```
>   "Attributes": {
>     "WinUsage": "08"
>   }
> ```
> 
> to match `\\\\?\\hid#vid_0458&pid_501a&col08#6&3e23d5&5&0007#{4d1e55b2-f16f-11cf-88cb-001111000030}` (col**08**)

but, would I need to wait for the PR to get rebased, or just clone the PR, build it and test it locally?

InfinityGhost, 2020-12-09:
> I followed the first step of adding the config and the tablet gets picked up, but as seen in the logs, it doesn't connect. Now, I assume following this comment is what I need to do
>
>
> but, would I [...]

fwiw you should be able to just use the [CI build](https://github.com/InfinityGhost/OpenTabletDriver/actions/runs/348464123)

vladfrangu, 2020-12-09:
Alright, I got the CI build running, and I took the JSON from the mentioned ZIP in this thread; changed the WinUsage to every one that matched the tablet - some wouldn't connect, some would; but no change was applied, with an area set to a quarter. Am I missing something? Do you need any further logs, or tests? I'm willing to help out to get support for this tablet! 😄

Mrcubix, 2021-10-11:
@vladfrangu it has been a while and WinUSB support has been implemented and is available in a pre-release (and master but don't use that) that's only available on discord right now, if you're down, you can join the [discord server](https://discord.gg/P3GYdUS) as there will be more people present to get the info we need.

Once you're there, hop into one of the help channels (preferably help-windows if on windows) and create a new thread,
The name of the thread should be the name of the tablet, when i or anyone is around, they will probably ping you to do stuff and get infos about the device.