OpenTabletDriver/OpenTabletDriver #405: Does not work with Gaomon S620

raf060708, 2020-10-28:
## Description

Tablet is not detected

## System Information:

Option            | Value
------------------|------------------
Operating System  | Windows 10
Software Version  |Latest
Tablet            | Gaomon S620

raf060708, 2020-10-28:
[image]

raf060708, 2020-10-28:
Already tried deleting Gaomon HID in device manager but it still doesn't work

X9VoiD, 2020-10-28:
Make sure you deleted the driver files when prompted, and also searched under **USB Controllers** for the device. When you're sure that they're gone, replug tablet, then restart OTD.

raf060708, 2020-10-28:
[image]
which one should I delete?

raf060708, 2020-10-28:
Sorry I am shit at this

X9VoiD, 2020-10-28:
Make sure you are using the correct view: `View -> Devices by Connection`

raf060708, 2020-10-28:
Which one should I delete
[image]

X9VoiD, 2020-10-28:
Follow the FAQ instructions. Open ACPI x64, find the USB controllers, then recursively search for the Gaomon HID, or whatever you'll find. Please don't ask step-by-step.

raf060708, 2020-10-28:
Deleted Gaomon tablet, It's all I could find and Tablet is still not detected

Mrcubix, 2020-10-28:
just replug your tablet now that you uninstalled AND removed the driver

raf060708, 2020-10-28:
Already did that and it still says that No Tablet detected

X9VoiD, 2020-10-28:
OTD -> Help -> Export Diagnostics, send the file here

Mrcubix, 2020-10-28:
also "Gaomon Tablet" might be there instead of "Gaomon HID" or just "HID Pen" because you might still have gaomon driver installed

raf060708, 2020-10-28:
[HEELEPME.txt](https://github.com/InfinityGhost/OpenTabletDriver/files/5452259/HEELEPME.txt)

X9VoiD, 2020-10-28:
According to the diagnostics, you still haven't done the instructions from FAQ correctly. Make sure you've uninstalled the Gaomon OEM Driver, and then perform the FAQ fix again, then restart OTD.

Mrcubix, 2020-10-28:
you still have gaomon driver installed, just confirmed by me now by installing the driver and checking my devices

raf060708, 2020-10-28:
Im fukcing dumb

raf060708, 2020-10-28:
[image]
i swear there is nothing

Mrcubix, 2020-10-28:
[image]

Mrcubix, 2020-10-28:
uninstall both of these is what i meant

raf060708, 2020-10-28:
okk

raf060708, 2020-10-28:
When I try to delete the second one this appears
[image]

Mrcubix, 2020-10-28:
reinstall gaomon driver, uninstall winusb first gaomon after

raf060708, 2020-10-28:
ok thx

raf060708, 2020-10-28:
Did all the steps but it still says no tablet detected

Mrcubix, 2020-10-28:
just to make sure, you uninstalled winUSB and gaomon drivers, you uninstalled vmulti if you ever installed it, you removed Gaomon HID in your devices with the following steps: 
- View > by connection
- search for the usb controller in ACPI x64, an look for Gaomon HID or Gaomon tablet

raf060708, 2020-10-28:
how do you  uninstall vmulti

Mrcubix, 2020-10-28:
in the UX: Help > export diagnostic and send it

raf060708, 2020-10-29:
[OpenTabletDriver.txt](https://github.com/InfinityGhost/OpenTabletDriver/files/5458660/OpenTabletDriver.txt)

raf060708, 2020-10-29:
Sorry it took a long time, Just woke up

raf060708, 2020-10-29:
?

raf060708, 2020-10-29:
hwlp

X9VoiD, 2020-10-29:
Join discord server for further hwlp: https://discord.gg/P3GYdUS

InfinityGhost, 2020-11-16:
Assuming this is resolved.

closed 2020-11-16 by InfinityGhost