OpenTabletDriver/OpenTabletDriver #407: Support for Wacom CTE-460

cl20103B11, 2020-10-29:
I only have this very old tablet :c

skyborgff, 2020-10-29:
Open OTD
Help -> export diagnostics and upload here

cl20103B11, 2020-10-29:
[diagnostics.txt](https://github.com/InfinityGhost/OpenTabletDriver/files/5458449/diagnostics.txt)

skyborgff, 2020-10-29:
tablet -> open configuration editor -> generate new configuration -> select your tablet. Do tell if it works btw

X9VoiD, 2020-10-29:
OTD in this case won't be able to properly generate a correct configuration since it had no access to report length information. Run both OTD Daemon and UX as sudo, and then export diagnostics again.

cl20103B11, 2020-10-30:
[test.txt](https://github.com/InfinityGhost/OpenTabletDriver/files/5463383/test.txt)

"Running as root and export diagnostics"

X9VoiD, 2020-10-30:
Use latest [dev build](https://github.com/InfinityGhost/OpenTabletDriver/actions/runs/336984458) (sign in to access download links) then download the zip below
> [CTE-460.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5465148/CTE-460.zip)

Extract content to `Configuration\Wacom`, please do note that the config contains dummy values for tablet dimensions and is only for the purpose of getting the tablet detected in OTD. The dimensions and specs will be updated later.

> [30-opentabletdriver.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5465194/30-opentabletdriver.zip)

Extract content to `/etc/udev/rules.d/`, then run `sudo udevadm control --reload-rules && sudo udevadm trigger`

X9VoiD, 2020-11-01:
Any news on this?

InfinityGhost, 2020-11-05:
@cl20103B11

cl20103B11, 2020-11-06:
Sorry for the late cuz too busy :c , It works, thx

X9VoiD, 2020-11-06:
Now that it's working, we should confirm the specs of the tablet.

Open OTD -> Tablet -> Tablet Debugger, then hover your pen slowly to bottom right most part of the tablet. Try to get the highest possible `Position` you can get.

And then check what `ReportID` it says when hovering pen, and when pen is out of range.

Lastly, check the maximum `Pressure` of the pen.

X9VoiD, 2020-11-11:
@cl20103B11 any news on the three values?

cl20103B11, 2020-11-21:
sorry I m too busy to do other stuff and I finally release myself. Idk why it can't detect my tablet now, when I look the console it say SetFeature failed

InfinityGhost, 2020-11-21:
Gotta love when someone just is like "woot it works" and ignores the entire issue after the fact. Assume the values are correct and we'll patch later if need be.

cl20103B11, 2020-11-25:
sorry about the late, I am too busy on study and finally finished all of the work recently. Highest Possible Position I get is (14760, 9225), ReportID is 72 and 64 when the cursor touch the edge, maximum pressure is 1023. 

Really sorry about very late reply orz

AbstractQbit, 2020-11-25:
Hmm, odd value of max Y coord looks a bit strange. Just to double-check, instead of trying to get both maxed out at once, can you try getting max values separately, first max X on the right edge and then max Y at the bottom edge of the tablet?

cl20103B11, 2020-11-26:
> Hmm, odd value of max Y coord looks a bit strange. Just to double-check, instead of trying to get both maxed out at once, can you try getting max values separately, first max X on the right edge and then max Y at the bottom edge of the tablet?

still got max X:14760 Y:9225

InfinityGhost, 2020-12-02:
Should have closed with #471, closing now.

closed 2020-12-02 by InfinityGhost