OpenTabletDriver/OpenTabletDriver #408: GAOMON S620 Not working on Ubuntu 20.04

DavidRamosSal, 2020-10-29:
## Description

I installed the software following the guide for Ubuntu, but when I open the GUI I get "No tablet detected".
The "Console" tab gives the following error in the detection process:
 [ArgumentOutOfRangeException:Error]	Value range is [0, 15]. (Parameter 'value')
 
I ran the udev rules script found in the FAQ successfully, but the situation in the GUI hasn't changed and the tablet is not being recognized.

## System Information:

Option            | Value
------------------|------------------
Operating System  | Ubuntu 20.04.1
Software Version  | OpenTabletDriver v0.4.1
Tablet            | GAOMON S620

X9VoiD, 2020-10-29:
Restart OTD Daemon

InfinityGhost, 2020-10-30:
```sh
systemctl --user restart opentabletdriver.service
```

robik65, 2020-11-01:
> systemctl --user restart opentabletdriver.service

Worked the at first, then I opened my pc the next day and the issue is till there and also this no longer works.
I am using the latest pop os build.

DavidRamosSal, 2020-11-01:
> ```shell
> systemctl --user restart opentabletdriver.service
> ```

Thanks for the reply. It didn't change anything. I wonder if I should try to install from scratch.

InfinityGhost, 2020-11-01:
Give reinstalling a try, and use `sudo apt install ./OpenTabletDriver.deb` to install it. Some apt frontends like gdebi don't properly install packages.

robik65, 2020-11-01:
> Give reinstalling a try, and use `sudo apt install ./OpenTabletDriver.deb` to install it. Some apt frontends like gdebi don't properly install packages.

This didn't fix it for me.

InfinityGhost, 2020-11-01:
Try the following commands, then replugging your tablet
```bash
echo 'KERNEL=="uinput", SUBSYSTEM=="misc", TAG+="uaccess", OPTIONS+="static_node=uinput"' | sudo tee -a /etc/udev/rules.d/99-custom.rules
sudo udevadm control --reload-rules
systemctl --user restart opentabletdriver.service
```

robik65, 2020-11-01:
> Try the following commands, then replugging your tablet
> 
> ```shell
> echo 'KERNEL=="uinput", SUBSYSTEM=="misc", TAG+="uaccess", OPTIONS+="static_node=uinput"' | sudo tee -a /etc/udev/rules.d/99-custom.rules
> sudo udevadm control --reload-rules
> systemctl --user restart opentabletdriver.service
> ```
Didn't work either

X9VoiD, 2020-11-01:
Make sure you've restarted OTD UX every time you restarted the OTD daemon. And just to be really sure:
```
sudo udevadm control --reload-rules && sudo udevadm trigger
```

robik65, 2020-11-01:
> sudo udevadm control --reload-rules && sudo udevadm trigger

I dont understand, but I tried that aswell and didn't work.

I also tried everyting you all posted above again and nothing worked

X9VoiD, 2020-11-01:
```
sudo rmmod hid_uclogic
sudo udevadm control --reload-rules && sudo udevadm trigger
systemctl --user restart opentabletdriver.service
```
then restart OTD UX

robik65, 2020-11-01:
> ```
> sudo rmmod hid_uclogic
> sudo udevadm control --reload-rules && sudo udevadm trigger
> systemctl --user restart opentabletdriver.service
> ```
> 
> then restart OTD UX

This fixed the issue for now. Thank you! <3

X9VoiD, 2020-11-01:
Restart your PC and see if the issue comes back.

robik65, 2020-11-01:
> Restart your PC and see if the issue comes back.

It came back

X9VoiD, 2020-11-01:
Run `lsmod` and see if hid_uclogic or wacom is loaded

robik65, 2020-11-01:
There is no wacom in there, but hid_uclogic has 0

X9VoiD, 2020-11-01:
Follow the [instructions here](https://github.com/InfinityGhost/OpenTabletDriver/wiki/Linux-FAQ#permanently-disable-the-wacom-module
) and instead of `wacom`, blacklist `hid_uclogic`.

Delete `/etc/udev/rules.d/30-opentabletdriver.rules`

Then [download this zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5471471/rules.zip):

And extract it as `/etc/udev/rules.d/99-opentabletdriver.rules`

Finally, restart PC

robik65, 2020-11-01:
> Follow the [instructions here](https://github.com/InfinityGhost/OpenTabletDriver/wiki/Linux-FAQ#permanently-disable-the-wacom-module) and instead of `wacom`, blacklist `hid_uclogic`.
>
> Delete [...]

After some trouble installing `dotnet-sdk`, at `./build.sh` the command returns this error 

_/usr/share/dotnet/sdk/3.1.403/Microsoft.Common.CurrentVersion.targets(1177,5): error MSB3644: The reference assemblies for .NETFramework,Version=v5.0 were not found. To resolve this, install the Developer Pack (SDK/Targeting Pack) for this framework version or retarget your application. You can download .NET Framework Developer Packs at https://aka.ms/msbuild/developerpacks [/home/robik65/OpenTabletDriver-udev/OpenTabletDriver.udev/OpenTabletDriver.udev.csproj]

The build failed. Fix the build errors and run again._

X9VoiD, 2020-11-01:
Just follow the instructions for "Permanently disable the wacom module", nothing more

robik65, 2020-11-01:
> Just follow the instructions 1 and 2 for "Permanently disable the wacom module", nothing more

Did that, it's fixed now. Thank you

DavidRamosSal, 2020-11-01:
> Follow the [instructions here](https://github.com/InfinityGhost/OpenTabletDriver/wiki/Linux-FAQ#permanently-disable-the-wacom-module) and instead of `wacom`, blacklist `hid_uclogic`.
>
> Delete [...]

Disabling `hid_uclogic` worked for me too, thanks for the help! Will close now.

closed 2020-11-01 by DavidRamosSal