OpenTabletDriver/OpenTabletDriver #412: Add support for Gaomon 1060 PRO

sbstratos79, 2020-10-31:
# Tablet Information

| Name            | Value |
| --------------- | ----- |
| Manufacturer    | Gaomon
| Product Name    | 1060 PRO
| Width           | 10 inches
| Height          | 6.25 inches
| Pressure Levels | 8192
| LPI             | 5080

## Diagnostic Information

[diagnostics.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5469756/diagnostics.zip)

X9VoiD, 2020-10-31:
Use latest [`ux dev build`](https://github.com/InfinityGhost/OpenTabletDriver/actions/runs/338759250) then Tablet -> Open device string reader

VendorID: 9580
ProductID: 109
String Index: 201

Copy and paste the device string to here

sbstratos79, 2020-10-31:
OEM02_T174_181125

X9VoiD, 2020-10-31:
[1060 Pro.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5469799/1060.Pro.zip)

Extract zip content to `Configurations\Gaomon`

Interestingly, this is a rebranded Huion New 1060 Plus model.

sbstratos79, 2020-10-31:
Thanks for the file. I'll check the config in the morning and report back. :) I actually tried copying the Huion New 1060 plus config for making it work for my tab. It didn't work so I figured it probably had something to do with the "Initialization String" option which I had left blank. Seems like that was the case.

sbstratos79, 2020-11-01:
Thanks! It works now. But only on the ux dev build. The installer version still doesn't detect it.

InfinityGhost, 2020-11-01:
> Thanks! It works now. But only on the ux dev build. The installer version still doesn't detect it.

It will be included in the next release.

closed 2020-11-03 by InfinityGhost

Hardyzz, 2020-12-05:
There seems to be two versions to the Gaomon 1060Pro. The one I have isn't recognized by the driver. Here is the diagnostics file. The specs are the same
[diagnostic.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5646565/diagnostic.zip)

X9VoiD, 2020-12-05:
@Hardyzz this is the exact same Gaomon 1060 Pro, uninstall the original huion driver along with the winusb driver to allow OTD to detect the tablet

sbstratos79, 2020-12-05:
As @X9VoiD had already said. There is only one 1060 PRO. The driver stopped detecting my tablet as well after about a week. X9Void guided me on discord. You have to uninstall the Huion Driver from the device manager under the "Devices by Container" window. Here's how you can find it. 
[image]

[image]

You'll see a Huion Tablet driver under the Huion Tablet container. Uninstall that. You can't see it in my Huion Tablet container, because I have already uninstalled it.