OpenTabletDriver/OpenTabletDriver #414: Support for pressure sensitivity in Absolute mode

sbstratos79, 2020-11-01:
## Description

I think the title is descriptive enough

AbstractQbit, 2020-11-01:
Pressure sensitivity is supported in "artist" modes, added by windows ink plugin. Regular modes are meant to provide minimal latency in games such as osu! (OTD is a replacement for hawku's TabletDriver, after all)

sbstratos79, 2020-11-01:
Yes. I am aware of that. But the reason I opened this issue was because of the other issue I opened requesting support for hotkeys in Windows Ink mode. Using paint programs without hotkeys is not desirable. 
So I opened this issue in case that feature can't be implemented. Having pressure sensitivity in absolute mode could've served as an alternative in that case.

skyborgff, 2020-11-01:
How would this work as an alternative?

sbstratos79, 2020-11-01:
Having pressure sensitivity in Absolute mode would be the same thing as Windows ink with hotkeys, wouldn't it? You could then use your tablet in painting programs.

skyborgff, 2020-11-01:
No, absolute mode doesn't have hotkeys.
Binding is the same regardless not the mode

sbstratos79, 2020-11-01:
I am quite sure it does. I just tested it and the hotkeys work in Absolute mode.

skyborgff, 2020-11-01:
Could you send a screenshot of your binding screen?

sbstratos79, 2020-11-01:
Apologies, it seems like Windows Ink also supports key binds. I could've sworn it didn't work when I tried yesterday. Most likely some kind of an error on my part.

sbstratos79, 2020-11-01:
Now I understand why it didn't work yesterday. I set the key to Right Click. Right Click works in absolute mode but not in Windows Ink.

X9VoiD, 2020-11-01:
This issue is a `wontfix` issue and should be closed. AbsoluteMode is **not** supposed to have any pressure sensitivity at all.

sbstratos79, 2020-11-01:
I agree. I will close the issue.

closed 2020-11-01 by sbstratos79