OpenTabletDriver/OpenTabletDriver #415: Support for custom key binds with Windows Ink mode plugin

sbstratos79, 2020-11-01:
## Description

Windows Ink feature doesn't let the user use the key binds. I think the mode becomes almost useless without the keybinds. Having to scroll using a scrollbar or zoom in with Ctrl++ instead of Ctrl+Pan in a painting program is way too tedious.

skyborgff, 2020-11-01:
That is a planned feature, for now you can use the script runner from VoidPlugins and AutoHotkey to have advanced multikey binding

sbstratos79, 2020-11-01:
Is there a tutorial for scriptrunner and Autohotkey for setting custom keybinds? I only have a rudimentary understanding of these things.

skyborgff, 2020-11-01:
AutoHotkey you will have to search yourself. For script runner just install the plugin like you did to windows ink, right-click the bingeing in OTD that you want to set, and select "Script Runner" in the first drop-down, and the script number in the second. In the tools tab put the path to the .AHK in the number "slot" you want

sbstratos79, 2020-11-01:
I looked for Script Runner in the releases but couldn't find it. I downloaded the zip from Void Plugins master branch but the Script runner folder has cs and csproj files instead of dlls. I copied them to the Plugins folder but they aren't installed. Is there any way to install it?

skyborgff, 2020-11-01:
Download from here as it won't be built until next release 

https://cdn.discordapp.com/attachments/628651971267788800/766729148898607124/ScriptRunner.dll

sbstratos79, 2020-11-01:
Thanks! It's installed now.

skyborgff, 2020-11-01:
Looking at #414 i misunderstood what you wanted. I thought you wanted multiple key hotkeys. That's not natively possible for now. One key binding is

sbstratos79, 2020-11-01:
I actually meant both. Keybindings support as well as keybindings combinations support. Although I was partially mistaken on the keybindings support part. While Windows Ink does support key binds, it doesn't seem to support all. It doesn't support Right Click for example. I haven't tested enough to find out what other keys it doesn't support.

skyborgff, 2020-11-01:
It doesn't because of a design decision that I don't agree with tbh. It "highjack" the 3 mouse buttons and maps them like if they were a pen. Left is click, right is the barrel button on a surface pen, and middle is eraser. You can not change this behaviour as it's hardcore's in the plugin. Giving that now I have a real case we're that is an issue, I'll try to talk them over changing it.

skyborgff, 2020-11-01:
For the keybindings combination for now use AHK. It's planned to be natively supported tho

sbstratos79, 2020-11-01:
Thanks. I am watching some AutoHotkey and AHKHID tutorials for enabling this. I'll check if it works with right click or not and report back.

AbstractQbit, 2020-11-01:
That hijacking is needed because your windows ink pointer is not your mouse pointer, so clicking mouse buttons would click at the last mouse location, not where the tablet points to

sbstratos79, 2020-11-01:
The AHK script works with Right Click.

X9VoiD, 2020-11-01:
WindowsInk cursor is **not** the same as the mouse cursor as AbstractQBit have explained. Triggering the "normal" left, right, and middle click event while WindowsInk is active **will cause issues**.

We can easily just fix the "hijacking" by providing new `IBinding` dedicated to WindowsInk and LinuxArtistMode but that will cause issues when a user left the pen bindings as the "normal" left, right, middle clicks. They are just **not** available from the devices that both output modes emulate. It will also become really tedious when switching from one output mode to another, you must change the bindings constantly or OTD will crash (null reference).

And also, you can just press space then drag to move around the canvas on most painting programs (krita, photoshop, etc)

Hence why this is another `wontfix` issue and should be closed.

X9VoiD, 2020-11-01:
A future change will come to address confusion for OTD's Left, Right and Middle binding's adaptive behaviour (which you guys call hijacking)

InfinityGhost, 2020-11-01:
This is not focused on this repository and should be moved to the Windows Ink repository source (https://github.com/X9VoiD/VoiDPlugins)

closed 2020-11-01 by InfinityGhost

sbstratos79, 2020-11-02:
Alright. I'll open another issue in the other repository