OpenTabletDriver/OpenTabletDriver #418: Finger/Touchpad support for Wacom CTH-470 on Windows 10

Espionage724, 2020-11-03:
I'd like to be able to switch between pen and touchpad modes on-the-fly on Windows 10. I'd like the generic touchpad gestures as well (2-finger vertical scroll, 2-finger tap-to-right-click). This seemingly works fine on Linux (Fedora 33; touchpad with gestures and pen support).

Currently, if I use OpenTabletDriver on Windows 10, only the pen works.

InfinityGhost, 2021-04-21:
An update on this, we have touch report parsing on Wacom tablets now but no way to bind them to anything. We want to test more tablets with touch support before we make any decisions on how that data will be handled.

gonX, 2021-10-22:
It's unfortunately on the 0.7.x milestone, and since 0.6.x is yet to release that means it likely won't be looked at in the near future unless someone sends in a valid PR.

Generally, you can assume that any open issues on the project aren't forgotten, especially if they have a milestone attached to them.

Mrcubix, 2024-07-21:
Update: As mentioned in #1664 Touch support is available via the `Enhanced Output Modes` plugin in Plugins > Plugin Manager, comes with customizable settings for each tablets in the `Filters` tab under `Touch Settings`.

Non-Native Touch gestures are also available via the `Touch Gestures` plugin, some may not work as expected so feedback is appreciated.

[image]

This unfortunately do not have a 2 finger to scroll gesture, but as an alternative, you can define specific part of your tablet to perform some actions supported by OpenTabletDriver & other Binding plugins.

For example, you could have a Pan Up & Down Gesture located on the side of the tablet like this :

[image]

And have it linked to something like changing the volume or scrolling : 

[image]

Finally, i want to say that supporting native gesture will probably not be possible properly until a PR implementing contact size & / or confidence are implemented in the touch reports. Otherwise, i am still not sure to this day how to do without.