OpenTabletDriver/OpenTabletDriver #421: Support Request: XP-PEN DECO 02

lmdfmh, 2020-11-04:
After installing the linux driver (aur), it stops working, does not respond to movement and buttons.
 In the debug window, everything works fine, all buttons are recognized and movement too.
[XP-PEN DECO 02.txt](https://github.com/InfinityGhost/OpenTabletDriver/files/5489013/XP-PEN.DECO.02.txt)

X9VoiD, 2020-11-04:
Check Tablet Debugger and see what ReportID it says during hover, and when pen is out of range

lmdfmh, 2020-11-04:
while hover ReportID:80, when out of range ReportID:96 also pen have eraser on other side ReportID:84

X9VoiD, 2020-11-04:
Hover on the bottom right most part of the physical tablet area and try to retrieve the highest possible value for `<Position>`

lmdfmh, 2020-11-04:
22352, 13970

X9VoiD, 2020-11-04:
Are you entirely sure now with those values? If yes, then

[Deco 02.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5489455/Deco.02.zip)

Extract zip contents to `Configurations\XP-Pen` then restart OTD

lmdfmh, 2020-11-04:
i checked values 10 times. yes i'm sure.
did it but no changes, still work only in debug window:(

X9VoiD, 2020-11-04:
Are you sure that you've replaced the config that is already in there? And restarted OTD? If yes then, 

[Deco 02.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5489553/Deco.02.zip)

X9VoiD, 2020-11-04:
Restart PC

lmdfmh, 2020-11-04:
i did. also replugged a tablet

X9VoiD, 2020-11-04:
Export Diagnostics

lmdfmh, 2020-11-04:
[deco02.txt](https://github.com/InfinityGhost/OpenTabletDriver/files/5489624/deco02.txt)

X9VoiD, 2020-11-04:
```
sudo udevadm control --reload-rules && sudo udevadm trigger
systemctl --user restart opentabletdriver
```
and then start UX

X9VoiD, 2020-11-04:
If still not working after that then, `stat /dev/uinput`

lmdfmh, 2020-11-04:
[inzane@archlinux ~]$ stat /dev/uinput
  Файл: /dev/uinput
  Размер: 0             Блоков: 0          Блок В/В: 4096   символьный специальный файл
Устройство: 5h/5d       Инода: 332         Ссылки: 1     Тип устройства: a,df
Доступ: (0600/crw-------)  Uid: (    0/    root)   Gid: (    0/    root)
Доступ:        2020-11-05 00:58:37.713321342 +0700
Модифицирован: 2020-11-05 00:58:37.713321342 +0700
Изменён:       2020-11-05 00:58:37.713321342 +0700
Создан:        -

X9VoiD, 2020-11-04:
You don't have proper udev rules. Download this [rule](https://github.com/InfinityGhost/OpenTabletDriver-udev/actions/runs/340468496)

Delete `/etc/udev/rules.d/30-opentabletdriver.rules`
Extract zip content as `/etc/udev/rules.d/99-opentabletdriver.rules`

Reboot

lmdfmh, 2020-11-04:
Pen works and pen binds also works but extra keys on tablet doesn't

X9VoiD, 2020-11-04:
You may do the following modification to the config to enable aux buttons:

```"ReportParser": "OpenTabletDriver.Vendors.XP_Pen.XP_PenReportParser"```

We're currently holding off on using the aux parsers since there will be a rework that will happen in the near future to support dynamic aux count. (Currently, OTD is hardcoded to support 6 aux buttons)

With that out of the way you may close this issue now.

lmdfmh, 2020-11-04:
thank you for your help

closed 2020-11-04 by lmdfmh