OpenTabletDriver/OpenTabletDriver #430: Support for UGEE M708 Tablet

ppalgankkoch, 2020-11-07:
I have an UGEE M708 tablet

InfinityGhost, 2020-11-07:
Please use the correct template, it helps us a lot when creating configurations.
[image]

We need the following:

- Width
- Height
- Pressure Levels
- LPI

We also need you to click in the menus

- Help -> Export Diagnostics

Then save the file and send it here.

ppalgankkoch, 2020-11-07:
Sorry. I could not see the templates (probably because I just didn't look hard enough).

Width: 254 mm
Length: 152.4 mm
Pressure Level: 0-2048
LPI: 5080

[diag.txt](https://github.com/InfinityGhost/OpenTabletDriver/files/5504974/diag.txt)

Mrcubix, 2020-11-07:
https://cdn.discordapp.com/attachments/628651971267788800/774680987400798238/UGEE_M708.json

this should be enough to detect the tablet at least, it will either work or it won't and your cursor won't move, but your area will appear in OTD, to initialize it, you'll need to use the latest master devbuild: 

https://github.com/InfinityGhost/OpenTabletDriver/actions/runs/350695372

go to tablet > string reader > 21827 in the first text field, 129 in the second field, 201 in the third field before the button, and then hit that button and paste here what you get

ppalgankkoch, 2020-11-07:
Hello

When I try to put in the values you sent, it shows this error:
[image]
Translated to English: Error
Error: The calling thread cannot access this object because it is owned by another thread.
Any way to fix this ?

Mrcubix, 2020-11-07:
alright, string reader in dev build is somehow busted so let's do it the old way, download 
[getstring-dump.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5505196/getstring-dump.zip)
also download and install https://dotnet.microsoft.com/download/dotnet-core/current/runtime
when you done installing .net core runtime 3.1, extract the dump zip file, start it, and wait until string 254 and then copy the whole console content here

ppalgankkoch, 2020-11-07:
Dump
VendorID:
21827
ProductID:
129
Found Device
GetString (1): UC-LOGIC
GetString (2): Digitizer
GetString (3): F401-HK708-STD
GetString (4): F401-HK708-STD
GetString (5): UC-LOIC
GetString (6): TABLET 1060
GetString (7): F401-HK708-STD
GetString (8): F401-HK708-STD
GetString (9): F401-HK708-STD
GetString (10): F401-HK708-STD
GetString (11): F401-HK708-STD
GetString (12): F401-HK708-STD
GetString (13): F401-HK708-STD
GetString (14): F401-HK708-STD
GetString (15): F401-HK708-STD
GetString (16): F401-HK708-STD
GetString (17): F401-HK708-STD
GetString (18): F401-HK708-STD
GetString (19): F401-HK708-STD
GetString (20): F401-HK708-STD
GetString (21): F401-HK708-STD
GetString (22): F401-HK708-STD
GetString (23): Test
GetString (24): F401-HK708-STD
GetString (25): F401-HK708-STD
GetString (26): F401-HK708-STD
GetString (27): F401-HK708-STD
GetString (28): F401-HK708-STD
GetString (29): F401-HK708-STD
GetString (30): F401-HK708-STD
GetString (31): F401-HK708-STD
GetString (32): F401-HK708-STD
GetString (33): F401-HK708-STD
GetString (34): F401-HK708-STD
GetString (35): F401-HK708-STD
GetString (36): F401-HK708-STD
GetString (37): F401-HK708-STD
GetString (38): F401-HK708-STD
GetString (39): F401-HK708-STD
GetString (40): F401-HK708-STD
GetString (41): F401-HK708-STD
GetString (42): F401-HK708-STD
GetString (43): F401-HK708-STD
GetString (44): F401-HK708-STD
GetString (45): F401-HK708-STD
GetString (46): F401-HK708-STD
GetString (47): F401-HK708-STD
GetString (48): F401-HK708-STD
GetString (49): F401-HK708-STD
GetString (50): F401-HK708-STD
GetString (51): F401-HK708-STD
GetString (52): F401-HK708-STD
GetString (53): F401-HK708-STD
GetString (54): F401-HK708-STD
GetString (55): F401-HK708-STD
GetString (56): F401-HK708-STD
GetString (57): F401-HK708-STD
GetString (58): F401-HK708-STD
GetString (59): F401-HK708-STD
GetString (60): F401-HK708-STD
GetString (61): F401-HK708-STD
GetString (62): F401-HK708-STD
GetString (63): F401-HK708-STD
GetString (64): F401-HK708-STD
[... 165 lines cut ...]
GetString (230): F401-HK708-STD
GetString (231): F401-HK708-STD
GetString (232): F401-HK708-STD
GetString (233): F401-HK708-STD
GetString (234): F401-HK708-STD
GetString (235): F401-HK708-STD
GetString (236): F401-HK708-STD
GetString (237): F401-HK708-STD
GetString (238): F401-HK708-STD
GetString (239): F401-HK708-STD
GetString (240): F401-HK708-STD
GetString (241): F401-HK708-STD
GetString (242): F401-HK708-STD
GetString (243): F401-HK708-STD
GetString (244): F401-HK708-STD
GetString (245): F401-HK708-STD
GetString (246): F401-HK708-STD
GetString (247): F401-HK708-STD
GetString (248): F401-HK708-STD
GetString (249): F401-HK708-STD
GetString (250): F401-HK708-STD
GetString (251): F401-HK708-STD
GetString (252): F401-HK708-STD
GetString (253): F401-HK708-STD
GetString (254): F401-HK708-STD

Mrcubix, 2020-11-07:
https://cdn.discordapp.com/attachments/628651971267788800/774691175532724264/UGEE_M708.json
put this in your config folder

ppalgankkoch, 2020-11-07:
[image]
Is it just fine here ?

Mrcubix, 2020-11-07:
nope, forgot to tell you to put it in the huion folder

ppalgankkoch, 2020-11-07:
Done.

Mrcubix, 2020-11-07:
just restart OTD and check if your tablet work now

ppalgankkoch, 2020-11-07:
[Detect:Info]	No tablet found.
[image]

Mrcubix, 2020-11-07:
go to tablet debugger, drag your pen around and tell me if values appear

ppalgankkoch, 2020-11-07:
Nothing appears; just blank.

Mrcubix, 2020-11-07:
https://cdn.discordapp.com/attachments/628651971267788800/774696911252226098/UGEE_M708.json

this config should work this time

ppalgankkoch, 2020-11-07:
I check the console and there is something:
[Detect:Warning]	More than 1 matching digitizer has been found.
[Device:Error]	Failed to open stream.
[DeviceIOException:Error]	Unable to open HID class device (\\?\hid#vid_5543&pid_0081&mi_01&col01#7&3d498ea&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}).
[NullReferenceException:Error]	Object reference not set to an instance of an object.
[NullReferenceException:Error]	Object reference not set to an instance of an object.
[Device:Error]	Failed to open stream.
[DeviceIOException:Error]	Unable to open HID class device (\\?\hid#vid_5543&pid_0081&mi_02&col01#7&1704b528&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\kbd).
[NullReferenceException:Error]	Object reference not set to an instance of an object.
[NullReferenceException:Error]	Object reference not set to an instance of an object.
[Device:Error]	Failed to open stream.
[DeviceIOException:Error]	Unable to open HID class device (\\?\hid#vid_5543&pid_0081&mi_00&col01#7&2c3f46b3&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}).
[NullReferenceException:Error]	Object reference not set to an instance of an object.
[Device:Error]	Failed to open stream.
[DeviceIOException:Error]	Unable to open HID class device (\\?\hid#vid_5543&pid_0081&mi_00&col02#7&2c3f46b3&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}).
[NullReferenceException:Error]	Object reference not set to an instance of an object.
[NullReferenceException:Error]	Object reference not set to an instance of an object.
[Device:Error]	Failed to open stream.
[DeviceIOException:Error]	Unable to open HID class device (\\?\hid#vid_5543&pid_0081&mi_01&col03#7&3d498ea&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}).
[NullReferenceException:Error]	Object reference not set to an instance of an object.
[NullReferenceException:Error]	Object reference not set to an instance of an object.

Does this help anything ?

Mrcubix, 2020-11-07:
so your tablet is detected, now i just have to figure out how to pen the stream

ppalgankkoch, 2020-11-07:
Okay.

ppalgankkoch, 2020-11-11:
just asking, any update ?

ppalgankkoch, 2020-11-21:
[image]
?

X9VoiD, 2020-11-21:
Do you mind if you elaborate more on that question mark?

ppalgankkoch, 2020-11-21:
My bad. Just didn't know what to type. The tablet area shows, and then I went and deleted the drivers that were originally with the tablet. Restarted OTD, area shows but when I put the pen on the tablet Windows Ink appears.

Mrcubix, 2020-11-21:
use 0.4.2, the process bug is fixed there

Mrcubix, 2020-11-21:
then follow https://youtu.be/35-Jz6_TZOc?t=536

ppalgankkoch, 2020-11-21:
[IOException:Error]	Can't write to this device.

Mrcubix, 2020-11-21:
oh i forgot your case is that the init string is 200

Mrcubix, 2020-11-21:
so the issue right now is first, determining the device string

ppalgankkoch, 2020-11-21:
[image]
Checked the Device String Reader again.
String: 鱀巀߿ྠ

Mrcubix, 2020-11-21:
for you to understand why i struggle making this tablet working is because a usual huion tablet dump is like this https://pastebin.com/KFxPn9aZ
you can see `GetString (200): ??O???`
and then the device string which is `OEM02_T18e_190814`

ppalgankkoch, 2020-11-21:
What line would I find the device string on ?

Mrcubix, 2020-11-21:
that is the question

ppalgankkoch, 2020-11-21:
Is there anyway to get a full dump instead of having to type what line it is every time
would getstring-dump work again ?

Mrcubix, 2020-11-21:
it would return the same thing as you can find above in the previosu message, but you can try again and paste it into a pastebin

ppalgankkoch, 2020-11-21:
getstring-dump just quits as soon as I open it

Mrcubix, 2020-11-21:
did you installed .net 5 and started the dump after extracting the zip file?

ppalgankkoch, 2020-11-21:
Yes, and it still crashes.

ppalgankkoch, 2020-11-21:
^ It crashed because I needed to use .net 3.1 instead of .net 5

Mrcubix, 2020-11-21:
the init string is indeed 200, for the device string try these:

- gonna update in 2s

ppalgankkoch, 2020-11-21:
[RegexParseException:Error]	Invalid pattern '?' at offset 1. Quantifier {x,y} following nothing.
[RegexParseException:Error]	Invalid pattern '?' at offset 1. Quantifier {x,y} following nothing.

Mrcubix, 2020-11-21:
try another one

ppalgankkoch, 2020-11-21:
"121": "DFS-UI F401-2011102" and "103": "F401-HK708-STD" come back with the same error: [IOException:Error]	Can't write to this device.

Mrcubix, 2020-11-21:
last try : `"3": "F401-HK708-STD"`

ppalgankkoch, 2020-11-21:
Nope, doesn't work. Sucks.
I have two config files (one made from "Generate configuration..." and the other one which is the one you made.
The console for the one file you made: https://pastebin.com/NUrYKRz5 vs the file which was generated through "Generate configuration...": https://pastebin.com/5aGV5TUT

Mrcubix, 2020-11-21:
alright, so we'll do a USB dump now, download this https://desowin.org/usbpcap/thankyou.html?file=1.5.4.0/USBPcapSetup-1.5.4.0.exe install it,
once that's done, close OTD and download the Manufacturer driver for your tablet, tell me when done

ppalgankkoch, 2020-11-21:
Done

Mrcubix, 2020-11-21:
alright, start manufacturer driver, then go to` C:\Program File\USBPcap` open `USBPcapCMD.exe`
once you started it, screenshot the black windows

ppalgankkoch, 2020-11-21:
[image]

Mrcubix, 2020-11-21:
alright, so type one, then type the name of the file you want then do these steps:

- move your pen for 7s
- unplug your tablet
- wait 3s
- replug your tablet
- move your pen for 7s
- press Q to quit

then upload the generated file located in the same folder here

ppalgankkoch, 2020-11-21:
Should I change the file extension to .txt ?

Mrcubix, 2020-11-21:
nope

ppalgankkoch, 2020-11-21:
https://www.filehosting.org/file/details/907610/uiFrxIPcRYsJPhAj/usbpcap.pcap

X9VoiD, 2020-11-22:
Updated configuration: [M708.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5579143/M708.zip)

Delete or move your other M708 configs to somewhere else then extract this as `Configurations\UGEE\M708.json`

This contains the exact initialization sequence from the oem driver. Might or might not work depending on which interface of the device HidSharp (a library that OTD uses) will see first.

ppalgankkoch, 2020-11-22:
[Detect:Info]	Searching for tablet 'UGEE M708'
[Detect:Warning]	More than 1 matching digitizer has been found.
[Detect:Debug]	Using device 'TABLET 1060'.
[Detect:Debug]	Using report parser type 'OpenTabletDriver.Tablet.TabletReportParser'.
[Detect:Debug]	Device path: \\?\hid#vid_5543&pid_0081&mi_01&col01#7&144dace4&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
[Device:Debug]	Initializing index 98
[Device:Debug]	Initializing index 122
[Device:Debug]	Initializing index 123
[Device:Debug]	Initializing index 102
[Device:Debug]	Initializing index 100
[Device:Debug]	Initializing index 101
[Device:Debug]	Initializing index 110
[Device:Error]	Failed to open stream.
[DeviceIOException:Error]	Unable to open HID class device (\\?\hid#vid_5543&pid_0081&mi_01&col01#7&144dace4&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}).
[Detect:Info]	Found tablet 'UGEE M708'

Mrcubix, 2020-11-22:
now go to tablet > tablet debugger
move your pen around and tell us if values change

ppalgankkoch, 2020-11-22:
It's blank

X9VoiD, 2020-11-22:
Download and setup [#425's dev build](https://github.com/InfinityGhost/OpenTabletDriver/actions/runs/348464123)
And then use this configuration instead: [M708.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5579570/M708.zip)

This will directly try to open the vendor interface

ppalgankkoch, 2020-11-22:
Same error

X9VoiD, 2020-11-22:
Send the log, it should atleast say now that it's trying to open `mi_00&col02`

ppalgankkoch, 2020-11-22:
[Detect:Info]	Searching for tablet 'UGEE M708'
[Detect:Warning]	More than 1 matching digitizer has been found.
[Detect:Debug]	Using device 'TABLET 1060'.
[Detect:Debug]	Using report parser type 'OpenTabletDriver.Tablet.TabletReportParser'.
[Detect:Debug]	Device path: \\?\hid#vid_5543&pid_0081&mi_00&col02#7&2c113929&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}
[Device:Debug]	Initializing index 98
[Device:Debug]	Initializing index 122
[Device:Debug]	Initializing index 123
[Device:Debug]	Initializing index 102
[Device:Debug]	Initializing index 100
[Device:Debug]	Initializing index 101
[Device:Debug]	Initializing index 110
[Device:Error]	Failed to open stream.
[DeviceIOException:Error]	Unable to open HID class device (\\?\hid#vid_5543&pid_0081&mi_00&col02#7&2c113929&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}).
[Detect:Info]	Found tablet 'UGEE M708'

X9VoiD, 2020-11-22:
More progress being made in HidSharpCore [#14](https://github.com/InfinityGhost/HIDSharpCore/pull/14). Also checking on the provided dumps for the possibility that it's using other methods for initialization.

[6 more replies]

closed 2021-12-28 by jamesbt365