OpenTabletDriver/OpenTabletDriver #432: Trouble getting Gaomon s260 detected by OTD.

notanosuplayer, 2020-11-08:
Hi, I'm having the same issue that I've seen others with currently. I uninstalled the gaomon tablet, HID tabler and HID pen from device manager, however I still can't get my tablet detected. Attached is my diagnostic file. I've read through the other issues, but nothing there was of too much help, so I figured it would be best to open a new one.
[gaomonsucks.txt](https://github.com/InfinityGhost/OpenTabletDriver/files/5506093/gaomonsucks.txt)

X9VoiD, 2020-11-08:
Open cmd prompt as admin, then type
```
pnputil -e
```
Send output here

notanosuplayer, 2020-11-08:
Microsoft Windows [Version 10.0.18362.1139]
(c) 2019 Microsoft Corporation. All rights reserved.

C:\Windows\system32>pnputil -e
Microsoft PnP Utility

Published name :            oem34.inf
Driver package provider :   Apple, Inc.
Class :                     Universal Serial Bus devices
Driver date and version :   05/07/2018 423.36.0.0
Signer name :               Microsoft Windows Hardware Compatibility Publisher

Published name :            oem3.inf
Driver package provider :   Intel
Class :                     System devices
Driver date and version :   07/07/2015 11.0.0.1157
Signer name :               Microsoft Windows Hardware Compatibility Publisher

Published name :            oem8.inf
Driver package provider :   Microsoft
Class :                     Human Interface Devices
Driver date and version :   10/27/2015 9.9.108.0
Signer name :               Microsoft Windows Hardware Compatibility Publisher

Published name :            oem2.inf
Driver package provider :   Intel
Class :                     Ports (COM & LPT)
Driver date and version :   05/08/2015 11.0.0.1136
Signer name :               Microsoft Windows Hardware Compatibility Publisher

Published name :            oem24.inf
Driver package provider :   NVIDIA
Class :                     Universal Serial Bus controllers
Driver date and version :   07/24/2019 1.1.27.831
Signer name :               Microsoft Windows Hardware Compatibility Publisher

Published name :            oem27.inf
Driver package provider :   NVIDIA
Class :                     Mice and other pointing devices
Driver date and version :   06/06/2018 8.16.23319.34391
Signer name :               Microsoft Windows Hardware Compatibility Publisher

Published name :            oem29.inf
Driver package provider :   NVIDIA
Class :                     System devices
Driver date and version :   01/10/2020 3.4.2785.9920
Signer name :               Microsoft Windows Hardware Compatibility Publisher

Published name :            oem28.inf
Driver package provider :   NVIDIA
Class :                     System devices
Driver date and version :   08/27/2018 3.3.2475.1519
Signer name :               Microsoft Windows Hardware Compatibility Publisher

Published name :            oem17.inf
Driver package provider :   REDRAGON
Class :                     Human Interface Devices
Driver date and version :   09/20/2017 1.0.1.0
Signer name :               Microsoft Windows Hardware Compatibility Publisher

Published name :            oem25.inf
Driver package provider :   SAMSUNG Electronics Co., Ltd.
Class :                     Universal Serial Bus controllers
Driver date and version :   04/21/2020 2.16.14.0
Signer name :               Microsoft Windows Hardware Compatibility Publisher

Published name :            oem37.inf
Driver package provider :   Unified Intents AB
Class :                     Human Interface Devices
Driver date and version :   08/04/2019 13.20.31.0
[... 146 lines cut ...]
Published name :            oem20.inf
Driver package provider :   VB-Audio Software
Class :                     Sound, video and game controllers
Driver date and version :   09/02/2014 1.0.3.5
Signer name :               Vincent Burel

Published name :            oem18.inf
Driver package provider :   GSPY
Class :                     GSPY Device
Driver date and version :   11/20/2013 1.0.0.0
Signer name :               Microsoft Windows Hardware Compatibility Publisher

Published name :            oem21.inf
Driver package provider :   Voicemod
Class :                     Sound, video and game controllers
Driver date and version :   01/10/2018 11.18.35.982
Signer name :               Microsoft Windows Hardware Compatibility Publisher

Published name :            oem12.inf
Driver package provider :   Western Digital Technologies
Class :                     WD Drive Management devices
Driver date and version :   11/30/2017 1.2.0.0
Signer name :               Microsoft Windows Hardware Compatibility Publisher

C:\Windows\system32>

X9VoiD, 2020-11-08:
```
pnputil -i -a %SystemRoot%/INF/input.inf
pnputil -e
```
And then upload result here again

notanosuplayer, 2020-11-08:
Microsoft Windows [Version 10.0.18362.1139]
(c) 2019 Microsoft Corporation. All rights reserved.

C:\Windows\system32>pnputil -e
Microsoft PnP Utility

Published name :            oem34.inf
Driver package provider :   Apple, Inc.
Class :                     Universal Serial Bus devices
Driver date and version :   05/07/2018 423.36.0.0
Signer name :               Microsoft Windows Hardware Compatibility Publisher

Published name :            oem3.inf
Driver package provider :   Intel
Class :                     System devices
Driver date and version :   07/07/2015 11.0.0.1157
Signer name :               Microsoft Windows Hardware Compatibility Publisher

Published name :            oem8.inf
Driver package provider :   Microsoft
Class :                     Human Interface Devices
Driver date and version :   10/27/2015 9.9.108.0
Signer name :               Microsoft Windows Hardware Compatibility Publisher

Published name :            oem2.inf
Driver package provider :   Intel
Class :                     Ports (COM & LPT)
Driver date and version :   05/08/2015 11.0.0.1136
Signer name :               Microsoft Windows Hardware Compatibility Publisher

Published name :            oem24.inf
Driver package provider :   NVIDIA
Class :                     Universal Serial Bus controllers
Driver date and version :   07/24/2019 1.1.27.831
Signer name :               Microsoft Windows Hardware Compatibility Publisher

Published name :            oem27.inf
Driver package provider :   NVIDIA
Class :                     Mice and other pointing devices
Driver date and version :   06/06/2018 8.16.23319.34391
Signer name :               Microsoft Windows Hardware Compatibility Publisher

Published name :            oem29.inf
Driver package provider :   NVIDIA
Class :                     System devices
Driver date and version :   01/10/2020 3.4.2785.9920
Signer name :               Microsoft Windows Hardware Compatibility Publisher

Published name :            oem28.inf
Driver package provider :   NVIDIA
Class :                     System devices
Driver date and version :   08/27/2018 3.3.2475.1519
Signer name :               Microsoft Windows Hardware Compatibility Publisher

Published name :            oem17.inf
Driver package provider :   REDRAGON
Class :                     Human Interface Devices
Driver date and version :   09/20/2017 1.0.1.0
Signer name :               Microsoft Windows Hardware Compatibility Publisher

Published name :            oem25.inf
Driver package provider :   SAMSUNG Electronics Co., Ltd.
Class :                     Universal Serial Bus controllers
Driver date and version :   04/21/2020 2.16.14.0
Signer name :               Microsoft Windows Hardware Compatibility Publisher

Published name :            oem37.inf
Driver package provider :   Unified Intents AB
Class :                     Human Interface Devices
Driver date and version :   08/04/2019 13.20.31.0
[... 642 lines cut ...]
Published name :            oem20.inf
Driver package provider :   VB-Audio Software
Class :                     Sound, video and game controllers
Driver date and version :   09/02/2014 1.0.3.5
Signer name :               Vincent Burel

Published name :            oem18.inf
Driver package provider :   GSPY
Class :                     GSPY Device
Driver date and version :   11/20/2013 1.0.0.0
Signer name :               Microsoft Windows Hardware Compatibility Publisher

Published name :            oem21.inf
Driver package provider :   Voicemod
Class :                     Sound, video and game controllers
Driver date and version :   01/10/2018 11.18.35.982
Signer name :               Microsoft Windows Hardware Compatibility Publisher

Published name :            oem12.inf
Driver package provider :   Western Digital Technologies
Class :                     WD Drive Management devices
Driver date and version :   11/30/2017 1.2.0.0
Signer name :               Microsoft Windows Hardware Compatibility Publisher

C:\Windows\system32>

X9VoiD, 2020-11-08:
Are you sure that your Gaomon S620 is plugged in right now?

notanosuplayer, 2020-11-08:
I am, the light is on and it responds to the pen

X9VoiD, 2020-11-08:
```
pnputil -f -d oem6.inf
```

X9VoiD, 2020-11-08:
```
pnputil -i -a %SystemRoot%/INF/input.inf
```
Then launch OTD

notanosuplayer, 2020-11-08:
[gaomonsucks(1).txt](https://github.com/InfinityGhost/OpenTabletDriver/files/5506140/gaomonsucks.1.txt)

X9VoiD, 2020-11-08:
There is **NOT** a single Gaomon S620 in your system right now unless you are using some weird drivers that hinder proper device detection by overriding the default ones.

Whatever driver you are using right now uninstall it first.

notanosuplayer, 2020-11-08:
As far as I know, I don't have any other tablet drivers installed or anything that would interact with my tablet in this way. This is very strange, but I've already uninstalled all the drivers and devices in device manager. Should I try another reboot and give you another diagnostic?

X9VoiD, 2020-11-08:
Yeah, might be worth a shot.

notanosuplayer, 2020-11-08:
[ilovegaomon.txt](https://github.com/InfinityGhost/OpenTabletDriver/files/5506182/ilovegaomon.txt)

X9VoiD, 2020-11-08:
Still the same, your Gaomon S620 is hidden behind a driver. Open Device Manager. Uninstalling anything that says Tablet

View -> Devices by Connection

Search under x86 ACPI -> (PCI Root if no immediate USB Controller) ->Any USB Controllers for anything that also resembles a tablet, uninstall them (and **delete** driver file when prompted)

Start OTD. If still not detected open Windows Settings -> Devices -> Mouse, Keyboard, and Pen -> Uninstalling anything that looks like a tablet.

And then restart OTD

notanosuplayer, 2020-11-08:
Checked again and did find one or two things in device manager that I missed. Uninstalled them, then went into windows settings and removed the tablet that showed up after replugging my s620. Still no dice with OTD. I may just have to go with the Gaomon drivers at this point.

X9VoiD, 2020-11-08:
Go with Gaomon drivers then and see if it installs fine. Because if it somehow found your device then OTD will be able to do so after killing gaomon.

X9VoiD, 2020-11-08:
When you uninstalled from Windows Settings, make sure that you replugged the tablet, since Windows "unplugs" the device when doing that method of uninstall

Mrcubix, 2020-11-08:
to me, it just look like you still have vmulti installed, so uninstall it

InfinityGhost, 2020-11-16:
Is this resolved?

closed 2020-12-02 by InfinityGhost