OpenTabletDriver/OpenTabletDriver #433: Cursor jumping to the last place my mouse was

foodcake, 2020-11-08:
I've installed the drivers and configured my tablet area, and every second or so my cursor will jump to the last place I used my mouse, I haven't touched anything.
CTL-6100WL
OpenTabletDriver Version 0.4.1
[image]
[image]
The 2nd image is a vague example of what I mean, I was circling my pen on the right and my cursor jumps to the top left for some reason, any theories?

X9VoiD, 2020-11-08:
Open Tablet -> Tablet Debugger, then check what ReportID it says when hovering pen and when pen is out of range

foodcake, 2020-11-08:
When still and hovering
[image]
When out of range
[image]

X9VoiD, 2020-11-08:
So I take it that the ReportID is a range between 16 and 19? Confirm what possible values ReportID can have and under what conditions.

foodcake, 2020-11-08:
Those are the only values I see, 16 when in range (whether dragging or hovering) and 19 when out of range. I do see the ReportID value flash for a split second, but none of the values on the left change when that happens, only ReportID flashes, and that's when my cursor jumps to the top.

X9VoiD, 2020-11-08:
Modify `Configurations\Wacom\CTL-6100WL.json` and change ActiveReportID to something like this:

Start: 16
StartInclusive: true
End: 19
EndInclusive: false

Then restart OTD

foodcake, 2020-11-08:
Thank you so much, that was the issue, have a great day!

closed 2020-11-08 by foodcake