OpenTabletDriver/OpenTabletDriver #437: Add support for

TheFantasticLoki, 2020-11-10:
# Tablet Information

| Name            | Star 06C |
| --------------- | ----- |
| Manufacturer    | XP-Pen |
| Product Name    | Star 06C |
| Width           | 10" |
| Height          | 6" |
| Pressure Levels | 8192 |
| LPI             | 5080 |

## Diagnostic Information

{
  "App Version": "OpenTabletDriver v0.4.2",
  "Operating System": {
    "Platform": 2,
    "ServicePack": "",
    "Version": "10.0.18363.0",
    "VersionString": "Microsoft Windows NT 10.0.18363.0"
  },
  "Environment Variables": {
    "USERDOMAIN_ROAMINGPROFILE": "DESKTOP-L0K1",
    "MOZ_CRASHREPORTER_STRINGS_OVERRIDE": "C:\\Users\\astey\\AppData\\Local\\Mozilla Firefox\\browser\\crashreporter-override.ini",
    "VBOX_HWVIRTEX_IGNORE_SVM_IN_USE": "1",
    "PROCESSOR_REVISION": "0802",
    "APPDATA": "C:\\Users\\astey\\AppData\\Roaming",
    "OS": "Windows_NT",
    "SystemDrive": "C:",
    "ProgramW6432": "C:\\Program Files",
    "CLASSPATH": "C:\\Program Files\\Java\\jre1.8.0_241\\bin",
    "Path": "C:\\Users\\astey\\AppData\\Local\\Mozilla Firefox;C:\\Program Files (x86)\\Common Files\\Intel\\Shared Libraries\\redist\\intel64\\compiler;C:\\Program Files\\Oculus\\Support\\oculus-runtime;C:\\Program Files (x86)\\Common Files\\Intel\\Shared Libraries\\redist\\intel64\\compiler;C:\\Program Files\\Java\\jre1.8.0_241\\bin;C:\\Program Files\\dotnet\\;C:\\Program Files\\nodejs\\;C:\\Program Files\\Java\\jre1.8.0_261\\bin;C:\\Users\\astey\\AppData\\Roaming\\npm;C:\\Users\\astey\\AppData\\Local\\GitHubDesktop\\bin;C:\\Users\\astey\\AppData\\Local\\Programs\\Microsoft VS Code\\bin;C:\\Windows\\system32;",
    "ComSpec": "C:\\Windows\\system32\\cmd.exe",
    "SNF": "C:\\ProgramData\\Snorlers\\snp.sc",
    "INTEL_DEV_REDIST": "C:\\Program Files (x86)\\Common Files\\Intel\\Shared Libraries\\",
    "__COMPAT_LAYER": "DetectorsMessageBoxErrors",
    "USERNAME": "astey",
    "TA_GAME_PATH": "F:\\Games\\SteamLibrary\\steamapps\\common\\Call of Duty Black Ops III\\",
    "CommonProgramFiles(x86)": "C:\\Program Files (x86)\\Common Files",
    "DriverData": "C:\\Windows\\System32\\Drivers\\DriverData",
    "OculusBase": "C:\\Program Files\\Oculus\\",
    "USERPROFILE": "C:\\Users\\astey",
    "ASW_SERVER_STATE": "1",
    "CommonProgramFiles": "C:\\Program Files\\Common Files",
    "VBOX_MSI_INSTALL_PATH": "C:\\Program Files\\Oracle\\VirtualBox\\",
    "NUMBER_OF_PROCESSORS": "16",
    "MOZ_CRASHREPORTER_PING_DIRECTORY": "C:\\Users\\astey\\AppData\\Roaming\\Mozilla\\Firefox\\Pending Pings",
    "MIC_LD_LIBRARY_PATH": "C:\\Program Files (x86)\\Common Files\\Intel\\Shared Libraries\\compiler\\lib\\mic",
    "COMPUTERNAME": "DESKTOP-L0K1",
    "SystemRoot": "C:\\Windows",
    "HOMEPATH": "\\Users\\astey",
    "SSLKEYLOGFILE": "\\\\.\\aswMonFltProxy\\FFFF9F84E01EFDA0",
    "TA_LOCAL_ASSET_CACHE": "F:\\Games\\SteamLibrary\\steamapps\\common\\Call of Duty Black Ops III\\share\\assetconvert\\",
    "ALLUSERSPROFILE": "C:\\ProgramData",
    "PSModulePath": "C:\\Program Files\\WindowsPowerShell\\Modules;C:\\Windows\\system32\\WindowsPowerShell\\v1.0\\Modules",
    "OneDrive": "C:\\Users\\astey\\OneDrive",
    "SESSIONNAME": "Console",
    "windir": "C:\\Windows",
    "ProgramFiles": "C:\\Program Files",
    "TA_TOOLS_PATH": "F:\\Games\\SteamLibrary\\steamapps\\common\\Call of Duty Black Ops III\\",
    "MOZ_CRASHREPORTER_EVENTS_DIRECTORY": "C:\\Users\\astey\\AppData\\Roaming\\Mozilla\\Firefox\\Crash Reports\\events",
    "TMP": "C:\\Users\\astey\\AppData\\Local\\Temp",
    "LOGONSERVER": "\\\\DESKTOP-L0K1",
    "TEMP": "C:\\Users\\astey\\AppData\\Local\\Temp",
    "MOZ_CRASHREPORTER_RESTART_ARG_0": "C:\\Users\\astey\\AppData\\Local\\Mozilla Firefox\\firefox.exe",
    "ProgramFiles(x86)": "C:\\Program Files (x86)",
    "CommonProgramW6432": "C:\\Program Files\\Common Files",
    "PROCESSOR_IDENTIFIER": "AMD64 Family 23 Model 8 Stepping 2, AuthenticAMD",
    "PROCESSOR_ARCHITECTURE": "AMD64",
    "MOZ_CRASHREPORTER_DATA_DIRECTORY": "C:\\Users\\astey\\AppData\\Roaming\\Mozilla\\Firefox\\Crash Reports",
[... 1215 lines cut ...]
      "Time": "2020-11-10T16:54:41.5541392-07:00",
      "Group": "Detect",
      "Message": "Searching for tablet 'XP-Pen G640s'",
      "Level": 1
    },
    {
      "Time": "2020-11-10T16:54:41.5541461-07:00",
      "Group": "Detect",
      "Message": "Searching for tablet 'XP-Pen G960 v2'",
      "Level": 1
    },
    {
      "Time": "2020-11-10T16:54:41.5541531-07:00",
      "Group": "Detect",
      "Message": "Searching for tablet 'XP-Pen Star 03v2'",
      "Level": 1
    },
    {
      "Time": "2020-11-10T16:54:41.5541601-07:00",
      "Group": "Detect",
      "Message": "No tablet found.",
      "Level": 1
    }
  ]
}

TheFantasticLoki, 2020-11-10:
How long does it usually take to add a tablet?

InfinityGhost, 2020-11-11:
> How long does it usually take to add a tablet?

For tablets that cooperate, it can be a matter of minutes. In cases where we have to discover workarounds however, that can take a while. XP-Pen tablets typically have a history of sometimes being a pain to get working, however.

X9VoiD, 2020-11-11:
Download [Star 06C.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5520871/Star.06C.zip)

Extract as `Configurations\XP-Pen\Star 06C.json`. Restart OTD.

Please do note that this config contain dummy values for tablet dimension and sensor specs. Those will be updated later.

TheFantasticLoki, 2020-11-11:
How do I select 1 screen I think it's causing issues with trying to map to both screens AND is still using the normal pentablet driver area from xppen even when it's unloaded and force closed.

X9VoiD, 2020-11-11:
Uninstall XP Pen OEM Driver.

And right click on display area.

TheFantasticLoki, 2020-11-11:
[image]

X9VoiD, 2020-11-11:
As I've said uninstall XP Pen's OEM driver.

X9VoiD, 2020-11-11:
Once you are done uninstalling XP Pen's OEM driver, restart PC, then launch OTD.

Open OTD -> Tablet -> Tablet Debugger, then hover your pen slowly to bottom right most part of the tablet. Try to get the highest possible `Position` you can get.

And then check what `ReportID` it says when hovering pen, and when pen is out of range.

Lastly, check the maximum `Pressure` of the pen as measure from OTD (to be sure that it really is 8192)

Report all those values here so we can update the config with real values.

TheFantasticLoki, 2020-11-11:
Report ID: Hover: 80 OFR: 96
Max Pressure: 8191
Max Position range: 50800, 30480

TheFantasticLoki, 2020-11-11:
Still getting 2 cursor positions and not locking to 1 monitor even with area set properly and tablet area doesn't change.

TheFantasticLoki, 2020-11-11:
So I only get 2 cursors on monitor 1 but not 2

TheFantasticLoki, 2020-11-11:
I think it might be the default windows driver causing issues.

X9VoiD, 2020-11-11:
Have you already uninstalled the other drivers and restarted PC?  If yes then update your config with this one: [Star 06C.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5521097/Star.06C.zip)

Close OTD, replug tablet, then see if the cursor is a **dot**. If it is a dot then you have successfully uninstalled all other drivers that may interfere with OTD.

If it is a dot already, launch OTD with updated config, then check if the **dot** cursor becomes **normal** cursor.

TheFantasticLoki, 2020-11-11:
it still controls my cursor but I already uninstalled the pentablet software through the given so idk now I already looked through all processes, details, and services so

TheFantasticLoki, 2020-11-11:
Still 2 positions at once and still goes to main monitor after disabling windows ink as well

TheFantasticLoki, 2020-11-11:
I Did not mean to close this @InfinityGhost

TheFantasticLoki, 2020-11-11:
it was a missclick

X9VoiD, 2020-11-11:
Open cmd then run: `pnputil -e`

Send output here.

TheFantasticLoki, 2020-11-11:
Microsoft Windows [Version 10.0.18363.1139]
(c) 2019 Microsoft Corporation. All rights reserved.

C:\Windows\system32>pnputil -e
Microsoft PnP Utility

Published name :            oem22.inf
Driver package provider :   Advanced Micro Devices, Inc
Class :                     System devices
Driver date and version :   09/29/2019 2.2.0.121
Signer name :               Microsoft Windows Hardware Compatibility Publisher

Published name :            oem16.inf
Driver package provider :   Advanced Micro Devices, Inc
Class :                     System devices
Driver date and version :   05/17/2019 2.2.0.116
Signer name :               Microsoft Windows Hardware Compatibility Publisher

Published name :            oem20.inf
Driver package provider :   Advanced Micro Devices, Inc.
Class :                     System devices
Driver date and version :   04/02/2019 2.0.1.0
Signer name :               Microsoft Windows Hardware Compatibility Publisher

Published name :            oem4.inf
Driver package provider :   Advanced Micro Devices, Inc.
Class :                     System devices
Driver date and version :   03/14/2016 2.0.1.0
Signer name :               Microsoft Windows Hardware Compatibility Publisher

Published name :            oem5.inf
Driver package provider :   Advanced Micro Devices
Class :                     System devices
Driver date and version :   04/17/2018 1.0.0.55
Signer name :               Microsoft Windows Hardware Compatibility Publisher

Published name :            oem17.inf
Driver package provider :   Advanced Micro Devices
Class :                     System devices
Driver date and version :   07/24/2019 1.0.0.72
Signer name :               Microsoft Windows Hardware Compatibility Publisher

Published name :            oem3.inf
Driver package provider :   Advanced Micro Devices Inc.
Class :                     Security devices
Driver date and version :   06/02/2017 4.4.0.0
Signer name :               Microsoft Windows Hardware Compatibility Publisher

Published name :            oem19.inf
Driver package provider :   Advanced Micro Devices Inc.
Class :                     Security devices
Driver date and version :   06/19/2019 4.11.0.0
Signer name :               Microsoft Windows Hardware Compatibility Publisher

Published name :            oem26.inf
Driver package provider :   Corsair Components, Inc.
Class :                     Universal Serial Bus controllers
Driver date and version :   07/14/2017 3.3.0.0
Signer name :               Microsoft Windows Hardware Compatibility Publisher

Published name :            oem24.inf
Driver package provider :   Corsair
Class :                     System devices
Driver date and version :   06/30/2020 3.0.88.0
Signer name :               Microsoft Windows Hardware Compatibility Publisher

Published name :            oem23.inf
Driver package provider :   Corsair
Class :                     Human Interface Devices
Driver date and version :   06/30/2020 3.0.88.0
[... 278 lines cut ...]
Published name :            oem30.inf
Driver package provider :   Oracle Corporation
Class :                     Network adapters
Driver date and version :   07/10/2020 6.1.12.39181
Signer name :               Oracle Corporation

Published name :            oem31.inf
Driver package provider :   Oracle Corporation
Class :                     Network Service
Driver date and version :   07/10/2020 6.1.12.39181
Signer name :               Oracle Corporation

Published name :            oem60.inf
Driver package provider :   VB-Audio Software
Class :                     Sound, video and game controllers
Driver date and version :   01/11/2019 2.1.5.2
Signer name :               Microsoft Windows Hardware Compatibility Publisher

Published name :            oem39.inf
Driver package provider :   VB-Audio Software
Class :                     Sound, video and game controllers
Driver date and version :   01/11/2019 2.1.5.2
Signer name :               Microsoft Windows Hardware Compatibility Publisher

C:\Windows\system32>

TheFantasticLoki, 2020-11-11:
???

X9VoiD, 2020-11-13:
Windows Ink cannot be disabled by any means, as it is a kernel level driver. The thing you've disabled is Windows Ink Workspace.

Windows Ink's activity is signified by a dot/diamond cursor. Verify that this is what you're seeing when OTD is not running and then we'll continue. If it's a normal cursor then you have another tablet driver running in the background.

TheFantasticLoki, 2020-11-19:
But I have not installed any other drivers and after looking extensively could not find anything tablet or pen related.

X9VoiD, 2020-11-19:
So you're seeing a normal cursor instead of a dot/diamond cursor?

TheFantasticLoki, 2020-11-20:
Yes and that's how it's always been. never had the dot. it was plug n play other than pentablet software for configuring draw area.

X9VoiD, 2020-11-20:
As long as you have the normal cursor then it means you still have a driver running in the background. All XP-Pens have Windows Ink mode as their default mode (when running driverless) so you **have** to see a dot cursor instead of the normal one so there won't be any conflicts with OTD's function.

OTD will automatically switch the tablet to vendor interface mode (normal cursor)

TheFantasticLoki, 2020-11-20:
Then how can I find the driver it is using cause like I said I spent a  lot of time looking and couldn't find anything

olekhova, 2020-12-13:
Hello, I have XP-Pen Star 06 (which is wireless variant of 06C).
Device ID is 28bd:0078, so I've adjusted settings  VendorID/ProductID to 
```
      "VendorID": 10429,
      "ProductID": 120,
```
It says tablet is detected, but nothing else. It does not react to pen movement or pen buttons.
It reacted to buttons before I run OTD, but after that - no more.
I need to replug (unpower) the receiver to get it back to not-vendor interface.

I did not install OEM drivers. In not-vendor interface the tablet is working, but as separate pointer (not a mouse pointer). I do not see the **dot** cursor though. Maybe because my monitor's resolution is too fine.
Without OTD it works in absolute mode without pressure support. And buttons on pad work too.

X9VoiD, 2020-12-13:
@olekhova OTD -> Help -> Export Diagnostics

olekhov, 2020-12-13:
Sorry, wrong account.
(Star06.json)[https://gist.github.com/olekhov/da507e56b21f3107461494694f24034c]

X9VoiD, 2020-12-13:
Open Tablet -> Tablet Debugger, and then see what ReportID it says when pen is in range, and when pen is out of range.

olekhov, 2020-12-13:
The window stays empty.

X9VoiD, 2020-12-13:
Upload your config here

olekhov, 2020-12-13:
This is yours Star06C.json up this thread with ProductID adjusted to 0x78.
```
{
  "Name": "XP-PEN STAR 06",
  "DigitizerIdentifiers": [
    {
      "Width": 260.0,
      "Height": 170.0,
      "MaxX": 52000.0,
      "MaxY": 34000.0,
      "MaxPressure": 8191,
      "ActiveReportID": {
        "Start": 80,
        "StartInclusive": true,
        "End": 96,
        "EndInclusive": false
      },
      "VendorID": 10429,
      "ProductID": 120,
      "InputReportLength": 8,
      "OutputReportLength": 8,
      "ReportParser": null,
      "FeatureInitReport": null,
      "OutputInitReport": "ArAE",
      "DeviceStrings": {},
      "InitializationStrings": []
    }
  ],
  "AuxilaryDeviceIdentifiers": [],
  "Attributes": {
    "libinputoverride": "1"
  }
}
```

X9VoiD, 2020-12-13:
Change OutputInitReport to ArAC

olekhov, 2020-12-13:
Well, now it has something.
This is when pen is out:
[image]
When pen is on tablet, ReporterID is 80

olekhov, 2020-12-14:
Pen works, but no pressure.
Buttons do not work, wheel does not no work.
```
Buttons:
1 2
3 4
5 6
```
Data for buttons, all have ReportID: 120
```
1: Position [1,0]   Raw: 02 F0 01 00 00 00 00 xx
2: Position [2,0]   Raw: 02 F0 02 00 00 00 00 xx
3: Position [4,0]   Raw: 02 F0 04 00 00 00 00 xx
4: Position [8,0]   Raw: 02 F0 08 00 00 00 00 xx
5: Position [16,0]  Raw: 02 F0 10 00 00 00 00 xx
6: Position [32,0]  Raw: 02 F0 20 00 00 00 00 xx
```
As for wheel, it reports the last byte in raw data, cycling 00 -> 40 -> C0 -> 80 when turning clockwise and
00 -> 80 -> C0 -> 40 when turning counter-clockwise. 
Wheel position is recognized as pressure.

olekhov, 2020-12-14:
raw data for pen movement, if you need it:
```
02 A0 xx xx yy yy pp pp
```
A0 - when pen is on tablet, not touching, and tablet detects it.
A1 - when tip is pressed.
C0 - when pen is out of tablet - it is not detected.
xx xx yy yy - coords, LSB, max values:
```
x: 0xC670 = 50800
y: 0x7710 = 30480
```
pp pp - is the pressure, LSB, max value is 0x1FFF = 8191.
lowest half of second byte in raw data is the pen button state.
```
A0 - pen is detected, hovering on the tablet
A1 - pen tip is pressed
A2 - low pen button is pressed
A4 - high pen is pressed
A3 - tip & low button
A5 - tip & high button
```

AbstractQbit, 2020-12-14:
Regular report parser should read this just fine https://github.com/InfinityGhost/OpenTabletDriver/blob/e52003e619a9f6be40ecb7e5f75136565ef7f795/OpenTabletDriver/Tablet/TabletReport.cs#L9-L24

olekhov, 2020-12-14:
OTD reports pressure and coords in Tablet debugger window. But these are not translated to apps.
I've tested on Krita, FireAlpaca and MyPaint.
The apps only receive coords and tip pressed/released. No pressure is reported.

AbstractQbit, 2020-12-14:
For pressure support you will need a Windows Ink plugin https://github.com/X9VoiD/VoiDPlugins/releases

olekhov, 2020-12-14:
Ok, with that plugin pressure works.
How to map buttons and can I use wheel?

Mrcubix, 2020-12-14:
wheel isn't supported yet and you can't map buttons in artist mode because it use windows ink, you could try using ahk and script runner tho if you know a bit about coding in that language https://github.com/X9VoiD/VoiDPlugins/suites/1658155978/artifacts/30815494

X9VoiD, 2020-12-15:
@olekhov open OTD -> Tablet -> Tablet Debugger, hover your pen around the edges of the tablet and try to get the highest value for Position.

olekhov, 2020-12-15:
I wrote a small logger of raw commands.
```
maxx: 0000c670  50800
maxy: 00007710  30480
```

TheFantasticLoki, 2021-02-09:
after a very long time and many different attempts at fixing my original issue I have not found a solution or any other drivers on the system. Can I get some more help on this as I am starting to notice more input delay with the normal drivers

X9VoiD, 2021-02-09:
Download/Install [latest release](https://github.com/InfinityGhost/OpenTabletDriver/releases/tag/v0.5.1) and then export diagnostics using that release.

TheFantasticLoki, 2021-02-09:
I FOUND THE ISSUE WOOOOOOO! It was some driver windows was automatically installing that was defined as a normal mouse driver. But Now i have some other issues such as raw input  causing it to like cut my area in half. These problems are kind of hard to explain over this so if we could arrange a quick discord call that would be great. BUT THE DRIVER IS WORKING YAAAAAAYYYYY

TheFantasticLoki, 2021-02-09:
Apparently github doesn't support json format so here is the diagnostic output for you.

{
  "App Version": "OpenTabletDriver v0.5.1",
  "Operating System": {
    "Platform": 2,
    "ServicePack": "",
    "Version": "10.0.18363.0",
    "VersionString": "Microsoft Windows NT 10.0.18363.0"
  },
  "Environment Variables": {
    "TA_GAME_PATH": "F:\\Games\\SteamLibrary\\steamapps\\common\\Call of Duty Black Ops III\\",
    "Path": "C:\\Program Files (x86)\\Common Files\\Intel\\Shared Libraries\\redist\\intel64\\compiler;C:\\Program Files\\Oculus\\Support\\oculus-runtime;C:\\Program Files (x86)\\Common Files\\Intel\\Shared Libraries\\redist\\intel64\\compiler;C:\\Program Files\\Java\\jre1.8.0_241\\bin;C:\\Program Files\\dotnet\\;C:\\Program Files\\nodejs\\;C:\\Program Files\\Git\\cmd;C:\\Users\\astey\\AppData\\Local\\Programs\\Python\\Python39\\Scripts\\;C:\\Users\\astey\\AppData\\Local\\Programs\\Python\\Python39\\;C:\\Program Files\\Java\\jre1.8.0_261\\bin;C:\\Users\\astey\\AppData\\Roaming\\npm;C:\\Users\\astey\\AppData\\Local\\GitHubDesktop\\bin;C:\\Users\\astey\\AppData\\Local\\Programs\\Microsoft VS Code\\bin;C:\\Windows\\system32;",
    "OneDriveConsumer": "C:\\Users\\astey\\OneDrive",
    "ProgramFiles(x86)": "C:\\Program Files (x86)",
    "CommonProgramW6432": "C:\\Program Files\\Common Files",
    "OneDrive": "C:\\Users\\astey\\OneDrive",
    "NUMBER_OF_PROCESSORS": "16",
    "USERNAME": "astey",
    "PROCESSOR_REVISION": "0802",
    "FPS_BROWSER_USER_PROFILE_STRING": "Default",
    "LOCALAPPDATA": "C:\\Users\\astey\\AppData\\Local",
    "ComSpec": "C:\\Windows\\system32\\cmd.exe",
    "SystemDrive": "C:",
    "SNP": "https://%66%65%65%64.%68%65%6C%70%65%72%62%61%72.%63%6F%6D?publisher=APSFFreeSS&co=US&userid=e6aa3523-5b11-50f5-799e-8bcde9144128&searchtype=sc&installDate=05/04/2020&barcodeid=53724003&channelid=3&av=windows",
    "INTEL_DEV_REDIST": "C:\\Program Files (x86)\\Common Files\\Intel\\Shared Libraries\\",
    "TA_TOOLS_PATH": "F:\\Games\\SteamLibrary\\steamapps\\common\\Call of Duty Black Ops III\\",
    "ProgramFiles": "C:\\Program Files",
    "PROCESSOR_ARCHITECTURE": "AMD64",
    "DXSDK_DIR": "C:\\Program Files (x86)\\Microsoft DirectX SDK (June 2010)\\",
    "ProgramData": "C:\\ProgramData",
    "HOMEDRIVE": "C:",
    "TMP": "C:\\Users\\astey\\AppData\\Local\\Temp",
    "PSModulePath": "C:\\Program Files\\WindowsPowerShell\\Modules;C:\\Windows\\system32\\WindowsPowerShell\\v1.0\\Modules",
    "TEMP": "C:\\Users\\astey\\AppData\\Local\\Temp",
    "DriverData": "C:\\Windows\\System32\\Drivers\\DriverData",
    "CLASSPATH": "C:\\Program Files\\Java\\jre1.8.0_241\\bin",
    "LOGONSERVER": "\\\\DESKTOP-L0K1",
    "HOMEPATH": "\\Users\\astey",
    "APPDATA": "C:\\Users\\astey\\AppData\\Roaming",
    "PUBLIC": "C:\\Users\\Public",
    "ProgramW6432": "C:\\Program Files",
    "PATHEXT": ".COM;.EXE;.BAT;.CMD;.VBS;.VBE;.JS;.JSE;.WSF;.WSH;.MSC",
    "VBOX_HWVIRTEX_IGNORE_SVM_IN_USE": "1",
    "MIC_LD_LIBRARY_PATH": "C:\\Program Files (x86)\\Common Files\\Intel\\Shared Libraries\\compiler\\lib\\mic",
    "OculusBase": "C:\\Program Files\\Oculus\\",
    "USERDOMAIN": "DESKTOP-L0K1",
    "PROCESSOR_LEVEL": "23",
    "ALLUSERSPROFILE": "C:\\ProgramData",
    "COMPUTERNAME": "DESKTOP-L0K1",
    "CommonProgramFiles(x86)": "C:\\Program Files (x86)\\Common Files",
    "CommonProgramFiles": "C:\\Program Files\\Common Files",
    "SystemRoot": "C:\\Windows",
    "SNF": "C:\\ProgramData\\Snorlers\\snp.sc",
    "VBOX_MSI_INSTALL_PATH": "C:\\Program Files\\Oracle\\VirtualBox\\",
    "TA_LOCAL_ASSET_CACHE": "F:\\Games\\SteamLibrary\\steamapps\\common\\Call of Duty Black Ops III\\share\\assetconvert\\",
    "SESSIONNAME": "Console",
    "FPS_BROWSER_APP_PROFILE_STRING": "Internet Explorer",
    "PROCESSOR_IDENTIFIER": "AMD64 Family 23 Model 8 Stepping 2, AuthenticAMD",
    "windir": "C:\\Windows",
    "OS": "Windows_NT",
    "USERDOMAIN_ROAMINGPROFILE": "DESKTOP-L0K1",
    "USERPROFILE": "C:\\Users\\astey"
  },
  "HID Devices": [
    {
      "DevicePath": "\\\\?\\hid#vid_1b1c&pid_1b3e&mi_00&col03#8&26ac1cf&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "VendorID": 6940,
      "ProductID": 6974,
      "CanOpen": true,
[... 1699 lines cut ...]
      "Time": "2021-02-09T05:39:34.7405469-07:00",
      "Group": "Settings",
      "Message": "Tip Binding: [Mouse Button Binding: Left]@1%",
      "Level": 1
    },
    {
      "Time": "2021-02-09T05:39:34.7405536-07:00",
      "Group": "Settings",
      "Message": "Pen Bindings: [0, ], [1, ]",
      "Level": 1
    },
    {
      "Time": "2021-02-09T05:39:34.7405605-07:00",
      "Group": "Settings",
      "Message": "Express Key Bindings: [0, ], [1, ], [2, ], [3, ], [4, ], [5, ]",
      "Level": 1
    },
    {
      "Time": "2021-02-09T05:39:34.7405681-07:00",
      "Group": "Settings",
      "Message": "Driver is auto-enabled.",
      "Level": 1
    }
  ]
}

TheFantasticLoki, 2021-02-09:
So turning raw input on in osu causes the tablet area to be cut in half but not really. still registers pen input in cutoff section just cursor won't go above the equator of the screen, Very odd. Also No options for adjusting wheel bindings say adding a combo of ALT+ScrollUp/Down for quickly adjusting volume in game. Will report back with more info at a later time when I have had more time to mess around with the driver now that it is working.

TheFantasticLoki, 2021-02-09:
Area config seems off for the Star06C This is as far over as you can possibly go on the tablet before it stops registering just drop the width a bit and possibly send me over an updated config with less dummy values since I am still using the config sent at the beginning of this thread which is incomplete and very outdated by now.
[image]

TheFantasticLoki, 2021-02-09:
Height seems to be off by quite a bit as well.
[image]

X9VoiD, 2021-02-09:
Open Tablet Debugger and hover your pen to try to get the highest possible value for `Position`. It'll be on the bottom right of the tablet.

TheFantasticLoki, 2021-02-09:
I already did that for this tablet if you look above

TheFantasticLoki, 2021-02-09:
I rechecked still the same max values as before and even tried tweaking some settings around in the config not sure what to do tho

X9VoiD, 2021-02-09:
Send me the config that you currently have, just to make sure that we're on the same page when I edit the configuration

TheFantasticLoki, 2021-02-09:
[OTD.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5952274/OTD.zip)

X9VoiD, 2021-02-09:
I already have the same updated configuration as you do, the tablet visualizer have scaling issues so this is already fine if you don't have any problems with it anymore (besides the tablet visualizer)

TheFantasticLoki, 2021-02-09:
well the raw input issue and the wheel issue still persist

X9VoiD, 2021-02-09:
Wheel is not supported yet, and the raw input is not an issue, it's on [osu!s' side](https://github.com/InfinityGhost/OpenTabletDriver/wiki/Windows-FAQ#my-cursor-works-fine-on-the-desktop-but-not-in-osu)

TheFantasticLoki, 2021-02-09:
then why do I not have the issue with OEM drivers but with yours? That sounds like an issue relating to OTD

[7 more replies]

PR #764 (2021-02-23): Add Star 06C configuration
## Verification

[#437 comment](https://github.com/InfinityGhost/OpenTabletDriver/issues/437#issuecomment-775907352)

Fixes #437
closed 2021-02-23 by InfinityGhost