OpenTabletDriver/OpenTabletDriver #450: Plugin ActionAttribute function runs on UX instead of Daemon

X9VoiD, 2020-11-16:
## Description

Action buttons of `IFilter` and `ITool` plugins run their methods on UX instead of Daemon, causing issues when an implementer of a plugin wants the Daemon plugin to be affected in some way through the action buttons.

Bug discovered while trying to train a basic deep learning model. Basically, UX action button don't affect the plugin loaded in daemon in any way.

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Software Version | v0.4.2 -> master

InfinityGhost, 2020-11-16:
https://github.com/InfinityGhost/OpenTabletDriver/wiki/Plugins#relevant-attributes
Not necessarily a bug, we just don't have a way to communicate to the daemon what method to execute so we let the plugin do that. I'm probably going to be removing it anyway since its basically gone unused.

X9VoiD, 2021-05-24:
ActionAttribute is obsolete now, and so is this issue

closed 2021-05-24 by X9VoiD