OpenTabletDriver/OpenTabletDriver #451: Add support for Acepen AP 604

Ceebox, 2020-11-16:
This shouldn't be too hard.

Name | Star 06C
-- | --
Manufacturer | Acepen
Product Name | AP604
Width | 6"
Height | 4"
Pressure Levels | 8192
LPI

X9VoiD, 2020-11-16:
Please use the provided template for tablet configuration requests.

X9VoiD, 2020-11-17:
While tablet is connected, do `OTD -> Help -> Export Diagnostics`, then upload file here. We cannot add support to a new tablet without the diagnostics.

Mrcubix, 2020-12-12:
Any news on that?

closed 2020-12-13 by Ceebox

X9VoiD, 2020-12-13:
Why is this closed?

Ceebox, 2020-12-13:
I have attempted to reinstall OpenTabletDriver so I can obtain the required diagnostics. Unfortunately, it appears to corrupt my windows installation every time, and I do not think this is possible.

X9VoiD, 2020-12-13:
Corrupt your what? OTD doesn't even use admin permissions to "corrupt" Windows.

X9VoiD, 2020-12-13:
All support for other tablets were added the same way. A diagnostics output is required.

Ceebox, 2020-12-13:
When I install OTD, my Windows installation corrupts. I get a BSOD and then I am forced to reinstall to the last system restore point.

X9VoiD, 2020-12-13:
It is not even technically possible for OTD to cause "BSOD", are you sure what you're installing is OpenTabletDriver by InfinityGhost, and not TabletDriver by hawku/devocub?

X9VoiD, 2020-12-13:
Do be reminded, that OTD runs in user-mode, and *cannot* cause a kernel-level issue like BSoD. If you are having problems with setup, just use the installer provided in [latest release](https://github.com/InfinityGhost/OpenTabletDriver/releases).

Ceebox, 2020-12-13:
What do you mean? I run the installer, my pc crashes and blue screens. This _is_ and issue with OTD or the installer.

X9VoiD, 2020-12-13:
[This](https://github.com/InfinityGhost/OpenTabletDriver/releases/download/v0.4.2/OpenTabletDriver.win-x64.zip) is a portable version of OpenTabletDriver. And this is what's being downloaded by the installer. Extract this to a folder where you want to install OTD as long as it's not an admin folder.

Launch `OpenTabletDriver.UX.Wpf.exe`

AbstractQbit, 2020-12-13:
Don't run ANYTHING as admin, even installer

Ceebox, 2020-12-13:
It just crashes when I open it.

X9VoiD, 2020-12-13:
Restart your pc, go to %localappdata% and delete OpenTabletDriver, and then launch the portable again.

Ceebox, 2020-12-13:
Unfortunately, this still does not work.
I would like to help people using this driver with this tablet, however I am unable to get this to work.

X9VoiD, 2020-12-13:
Open Event Viewer -> Windows Logs -> Application.

Click on the latest log where it says `.NET Runtime`

Screenshot the contents of the log

Ceebox, 2020-12-13:
`Event Name: APPCRASH
Response: Not available
Cab Id: 0

Problem signature:
P1: OpenTabletDriver.Installer.exe
P2: 0.4.1.0
P3: 5f1757ad
P4: RTSSHooks64.dll
P5: 0.0.0.0
P6: 5c83708c
P7: c0000005
P8: 0000000000019fe9
P9: 
P10: 

Attached files:
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER310B.tmp.WERInternalMetadata.xml

These files may be available here:
\\?\C:\ProgramData\Microsoft\Windows\WER\ReportArchive\AppCrash_OpenTabletDriver_909ed87733f8928dbf68de569dbd48183b7f9653_306c3ef2_84aa3cb9-5c60-43aa-8891-e0d58227206b

Analysis symbol: 
Rechecking for solution: 0
Report Id: 2e19cd29-cdaa-4530-855c-5d375f13e073
Report Status: 268435456
Hashed bucket: 0c7744a2207e9e0271b4b6f1a330472d
Cab Guid: 0`

`Fault bucket , type 0
Event Name: APPCRASH
Response: Not available
Cab Id: 0

Problem signature:
P1: OpenTabletDriver.Installer.exe
P2: 0.4.1.0
P3: 5f1757ad
P4: RTSSHooks64.dll
P5: 0.0.0.0
P6: 5c83708c
P7: c0000005
P8: 0000000000019fe9
P9: 
P10: 

Attached files:

These files may be available here:
\\?\C:\ProgramData\Microsoft\Windows\WER\ReportQueue\AppCrash_OpenTabletDriver_909ed87733f8928dbf68de569dbd48183b7f9653_306c3ef2_84aa3cb9-5c60-43aa-8891-e0d58227206b

Analysis symbol: 
Rechecking for solution: 0
Report Id: 2e19cd29-cdaa-4530-855c-5d375f13e073
Report Status: 4
Hashed bucket: 
Cab Guid: 0`

There are two, so I posted both but I think they are the same.

AbstractQbit, 2020-12-13:
You are still launching the installer, use portable that X9VoiD sent you.

Ceebox, 2020-12-13:
Sorry, I have edited it. I copied the wrong one.

X9VoiD, 2020-12-13:
This is not the log that we need. The log should have a header looking something like:
```
Application: OpenTabletDriver.UX.Wpf.exe
CoreCLR Version: 5.0.20.51904
.NET Version: 5.0.0
Description: The process was terminated due to an unhandled exception.
Exception Info: Error something something
```

Ceebox, 2020-12-13:
Well, that is unfortunate, because all the logs I can find do not look like that.

[image]

X9VoiD, 2020-12-13:
From that log it's now clear what the problem is. RivaTuner is most likely overriding a default behaviour that OTD expects (through that RTSSHooks64.dll).

An OTD crash looks like this 
[image]

And the logs you've sent is an error external to OTD, since it did not came from CoreCLR (.NET Runtime)

Unfortunately, you *have* to uninstall RivaTuner and just use the official gpu drivers to tweak your stuff.

Ceebox, 2020-12-13:
Ah, that is unfortunate. Looks like I'm just going to have to go with the default settings and just try and adjust stuff in photoshop.