OpenTabletDriver/OpenTabletDriver #452: s620 problems (lubuntu)

wagwanfam, 2020-11-17:
## Description

tablet wont detect, also cant figure out how to save custom configurations so i cant try to use that to fix
## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System | lubuntu
| Software Version | v0.4.1 (because 0.4.2 wont install, cant get dot net 5 working)
| Tablet           | s620

wagwanfam, 2020-11-17:
"[Normal:Detect]	Searching for tablet 'Gaomon S620'
[Error:Detect]	Failed to get device input report length. Ensure the 'hid-uclogic' module is disabled."
^ message only appears after looking for s620 and some random huion tablets
so i did "sudo modprobe -r hid-uclogic"
and now i get  an error that says "[Error:HidSharp.Exceptions.DeviceIOException]	Unable to open HID class device (OK)."
no random googling helped me here :C

Mrcubix, 2020-11-17:
device string and or init string is wrong in your config

Mrcubix, 2020-11-17:
```
wget https://packages.microsoft.com/config/ubuntu/20.04/packages-microsoft-prod.deb -O packages-microsoft-prod.deb
sudo dpkg -i packages-microsoft-prod.deb
sudo apt-get update && \
sudo apt-get install -y dotnet-runtime-5.0
```
yoinked from void

wagwanfam, 2020-11-17:
right that fixed install
opened OTD 0.4.2 and still no detect
console reads 
"
[Detect:Info]	Searching for tablet 'Gaomon S620'
[DeviceIOException:Error]	Unable to open HID class device (OK).
"
iv wrote my own script now for my tablet to play osu!  but i wouldn't mind still helping resolve this issue in case others experience this

X9VoiD, 2020-11-17:
`sudo udevadm control --reload-rules && sudo udevadm trigger`

wagwanfam, 2020-11-17:
[ArgumentOutOfRangeException:Error]	Value range is [0, 15]. (Parameter 'value')
new error,(well first error i had b4 i was messing around)

X9VoiD, 2020-11-17:
Reboot

wagwanfam, 2020-11-17:
been out for exam came back, reeboted, same error

X9VoiD, 2020-11-18:
Blacklist `hid_uclogic` module

X9VoiD, 2020-11-19:
```
sudo rmmod hid_uclogic
sudo udevadm control --reload-rules && sudo udevadm trigger
systemctl --user restart opentabletdriver.service
```

closed 2020-12-02 by InfinityGhost