OpenTabletDriver/OpenTabletDriver #456: Cursor jumps to corner of the area

b4mbus88, 2020-11-19:
Hi,
iam using the Wacom PTH-660 and I cant use my mouse while the tablet is running. Wacom Drivers are uninstalled. 
Technicly the mouse is working but every ~2 seconds the cursor is jumping to the corner of the tablet area. 
If iam using the pen to move the cursor everything works fine.

There is some Warning and error too. Maybe it will help to solve the problem.

[Detect:Info]	Searching for tablet 'Wacom PTH-660 Bluetooth'
[Detect:Info]	Searching for tablet 'Wacom PTH-660'
[Detect:Info]	Found tablet 'Wacom PTH-660'
[Detect:Warning]	More than 1 matching auxiliary device has been found.
[Device:Error]	Failed to open stream.
[DeviceIOException:Error]	Unable to open HID class device (\\?\hid#vid_056a&pid_0357&mi_00&col01#7&25ebc169&1&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}).
[NullReferenceException:Error]	Object reference not set to an instance of an object.

X9VoiD, 2020-11-19:
Download [PTH-660.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5564117/PTH-660.zip) then extract and replace `Configurations/Wacom/PTH-660.json`.

After replacing, restart OTD then check again.

b4mbus88, 2020-11-19:
Still the same issue. As soon as i lay down my pen to use the mouse, the cursor is jumping to the corner every 2sec

X9VoiD, 2020-11-19:
Open Tablet Debugger then check what ReportID it says during hover, and when pen is not in range.

b4mbus88, 2020-11-19:
During hover
[image]
No hover
[image]

X9VoiD, 2020-11-19:
Is ReportID changing during hover based on distance?

b4mbus88, 2020-11-19:
Yes, ReportID is changeing to 19 as soon as i leave the tablet. I guess its around 2cm away from the real tablet area. dosent matter in which direction.

b4mbus88, 2020-11-19:
And the cursor is instantly jumping to the corner as soon ReportID is going to 19

X9VoiD, 2020-11-19:
Okay, I got what I need now, but just for verification, open Tablet Debugger again, then hover pen to bottom-right most part of the tablet and try to get the highest possible value for Position. Press down the pen to try to get the highest Pressure possible, then report those values here.

Also do Help -> Export Diagnostics then upload the file here. (to possibly fix the warnings)

b4mbus88, 2020-11-19:
Export Diagnostics
[diag.txt](https://github.com/InfinityGhost/OpenTabletDriver/files/5564224/diag.txt)
Tablet Debugger
[image]

Dont know if it is important, but i rotated the tablet area 180°

X9VoiD, 2020-11-19:
Get the highest value for Position

b4mbus88, 2020-11-19:
44800, 29600
Pressure 8191

Post above is edited with the picture

X9VoiD, 2020-11-19:
This [PTH-660.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5564261/PTH-660.zip) should fix the cursor teleport issue and the warning.

b4mbus88, 2020-11-19:
Sadly it still teleporting :( 
I remembered I used hawku/TabletDriver before. I uninstalled Wacom drivers but do i have to remove vmulti_drivers from hawku too?

X9VoiD, 2020-11-19:
Are you sure you replaced the existing configuration and restarted OTD?

b4mbus88, 2020-11-19:
Yes. I tried it again now, but it still dosent work.

b4mbus88, 2020-11-19:
Ohh.. If iam opening via Daemon it works. If iam using OpenTabletDriver.Installer.exe to start it is not working. Let me check if there is another folder

X9VoiD, 2020-11-19:
[PTH-660.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5564275/PTH-660.zip)

I changed the name just to make sure. Check on logs if it's detected as "PTH-660 Test"

b4mbus88, 2020-11-19:
Yeah. Iam stupid. Both of your last files work. There is just one installation folder in appdata and one in my downloads.
Thank you very much for helping fixing the problem!

X9VoiD, 2020-11-19:
You may close the issue now and no problemo :bread:

closed 2020-11-19 by b4mbus88