OpenTabletDriver/OpenTabletDriver #460: stuck on Connecting to OpenTabletDriver Daemon...

NoahDTheHorse, 2020-11-19:
## Description

sometimes when I open the OpenTabletDriver it just says "Connecting to OpenTabletDriver Daemon...", and will never load.
## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System |windows 10
| Software Version |0.4.2
| Tablet|Huion HS611

Mrcubix, 2020-11-19:
https://dotnet.microsoft.com/download/dotnet/thank-you/runtime-desktop-5.0.0-windows-x64-installer

NoahDTheHorse, 2020-11-19:
what is that?

Mrcubix, 2020-11-19:
a necessary dependency to make the UX work, btw a google search wouldn't hurt every once in a while https://youtu.be/A6JHLBiRRkQ

NoahDTheHorse, 2020-11-20:
I still had the problem after installing that, also I did not need to install it the way you did in the video, I was able  to download the exe directly and just run it that way. Also to fix the problem all I need to do is close all the openTabletDriver tasks in task manager and then reopen the program. I just thought I would post the bug

Mrcubix, 2020-11-20:
you minimized it, so it was hiding in your system tray, it's not a bug

NoahDTheHorse, 2020-11-20:
If the only way for me to reopen the gui is by running the program, and it does not work, I would classify that as a bug. I was not able to click on it from the taskbar like you are able to do when you minimize every other program

Mrcubix, 2020-11-20:
you said you had to kill the UX 

> all I need to do is close all the OpenTabletDriver tasks in task manager and then reopen the program.

by that what i understood was: you opened multiple UX, minimized one or more so they hid in your system tray, preventing you from launching another one, so just click on the OTD icon in the system tray and the UX will appear

closed 2020-12-02 by InfinityGhost