OpenTabletDriver/OpenTabletDriver #461: No Tablet Found

Ethn91081, 2020-11-20:
## Description

Hi, sorry to be one of those people but I tried to switch from the previous Infinity Ghost driver that supported the Gaomon S620 to this new one and it won't connect to my tablet. The previous driver was working fine. I'm relatively inept when it comes to stuff like this and I'm sorry to have been a burden
## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System Windows 10|
| Software Version 0.4.2|
| Tablet Gaomon S620|

Mrcubix, 2020-11-20:
uninstall vmulti if not already done ,then proceed with watching this vid 
https://www.youtube.com/watch?v=35-Jz6_TZOc

Ethn91081, 2020-11-20:
sorry to bother you again, but I was watching the video and in my device manager I don't have a Gaomon Tablet. Instead I have an HID-compliant pen and an HID-compliant-vendor-defined device

Mrcubix, 2020-11-20:
well then follow the rest of the vid after device uninstallation
don't forget to delete vmulti, manufacturer drivers according to the vid and all

closed 2020-12-02 by InfinityGhost