OpenTabletDriver/OpenTabletDriver #467: Support Request:

Jahquise, 2020-11-22:
when i open the driver it crashes i've tried restarting my pc and uninstalling and reinstalling

Mrcubix, 2020-11-23:
coincidentally, if you type opentabletdriver on google and check the video tab, on the third result, you can find this wonderful tutorial of mine https://youtu.be/A6JHLBiRRkQ

Jahquise, 2020-11-23:
i followed the video but it continues to crash it'll start to load up then crash

Mrcubix, 2020-11-23:
alright, go into event viewer, windows log > applications > click on the latest .net core runtime error you can find then send the content

Jahquise, 2020-11-23:
it was working this morning but now its's crashing

Jahquise, 2020-11-23:
Log Name:      Application
Source:        Application Error
Date:          11/22/2020 7:32:23 PM
Event ID:      1000
Task Category: (100)
Level:         Error
Keywords:      Classic
User:          N/A
Computer:      DESKTOP-S7THMED
Description:
Faulting application name: OpenTabletDriver.UX.Wpf.exe, version: 0.4.2.0, time stamp: 0x5f8de3c6
Faulting module name: KERNELBASE.dll, version: 10.0.19041.572, time stamp: 0x1183946c
Exception code: 0xe0434352
Fault offset: 0x0000000000023e49
Faulting process id: 0x12b8
Faulting application start time: 0x01d6c1301ab55f63
Faulting application path: C:\Users\Jahquise\AppData\Local\Temp\Rar$EXa11068.36723\OpenTabletDriver.UX.Wpf.exe
Faulting module path: C:\WINDOWS\System32\KERNELBASE.dll
Report Id: b003c6f8-54d7-49b0-b834-a8a8e35ad588
Faulting package full name: 
Faulting package-relative application ID: 
Event Xml:
<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
  <System>
    <Provider Name="Application Error" />
    <EventID Qualifiers="0">1000</EventID>
    <Version>0</Version>
    <Level>2</Level>
    <Task>100</Task>
    <Opcode>0</Opcode>
    <Keywords>0x80000000000000</Keywords>
    <TimeCreated SystemTime="2020-11-23T00:32:23.4002650Z" />
    <EventRecordID>7058</EventRecordID>
    <Correlation />
    <Execution ProcessID="0" ThreadID="0" />
    <Channel>Application</Channel>
    <Computer>DESKTOP-S7THMED</Computer>
    <Security />
  </System>
  <EventData>
    <Data>OpenTabletDriver.UX.Wpf.exe</Data>
    <Data>0.4.2.0</Data>
    <Data>5f8de3c6</Data>
    <Data>KERNELBASE.dll</Data>
    <Data>10.0.19041.572</Data>
    <Data>1183946c</Data>
    <Data>e0434352</Data>
    <Data>0000000000023e49</Data>
    <Data>12b8</Data>
    <Data>01d6c1301ab55f63</Data>
    <Data>C:\Users\Jahquise\AppData\Local\Temp\Rar$EXa11068.36723\OpenTabletDriver.UX.Wpf.exe</Data>
    <Data>C:\WINDOWS\System32\KERNELBASE.dll</Data>
    <Data>b003c6f8-54d7-49b0-b834-a8a8e35ad588</Data>
    <Data>
    </Data>
    <Data>
    </Data>
  </EventData>
</Event>

Mrcubix, 2020-11-23:
sorry but i'm looking for something like this
[image]

Mrcubix, 2020-11-23:
with .net core runtime once again

Jahquise, 2020-11-23:
Log Name:      Application
Source:        .NET Runtime
Date:          11/22/2020 7:32:23 PM
Event ID:      1026
Task Category: None
Level:         Error
Keywords:      Classic
User:          N/A
Computer:      DESKTOP-S7THMED
Description:
Application: OpenTabletDriver.UX.Wpf.exe
CoreCLR Version: 5.0.20.51904
.NET Version: 5.0.0
Description: The process was terminated due to an unhandled exception.
Exception Info: StreamJsonRpc.ConnectionLostException: The JSON-RPC connection with the remote party was lost before the request could complete.
   at StreamJsonRpc.JsonRpc.InvokeCoreAsync(JsonRpcRequest request, Type expectedResultType, CancellationToken cancellationToken)
   at StreamJsonRpc.JsonRpc.InvokeCoreAsync[TResult](RequestId id, String targetName, IReadOnlyList`1 arguments, IReadOnlyList`1 positionalArgumentDeclaredTypes, IReadOnlyDictionary`2 namedArgumentDeclaredTypes, CancellationToken cancellationToken, Boolean isParameterObject)
   at OpenTabletDriver.UX.MainForm.InitializeAsync() in C:\Users\Infinity\OpenTabletDriver\OpenTabletDriver.UX\MainForm.cs:line 687
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__140_0(Object state)
   at System.Windows.Threading.ExceptionWrapper.InternalRealCall(Delegate callback, Object args, Int32 numArgs)
   at System.Windows.Threading.ExceptionWrapper.TryCatchWhen(Object source, Delegate callback, Object args, Int32 numArgs, Delegate catchHandler)
   at System.Windows.Threading.DispatcherOperation.InvokeImpl()
   at MS.Internal.CulturePreservingExecutionContext.CallbackWrapper(Object obj)
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state)
--- End of stack trace from previous location ---
   at MS.Internal.CulturePreservingExecutionContext.Run(CulturePreservingExecutionContext executionContext, ContextCallback callback, Object state)
   at System.Windows.Threading.DispatcherOperation.Invoke()
   at System.Windows.Threading.Dispatcher.ProcessQueue()
   at System.Windows.Threading.Dispatcher.WndProcHook(IntPtr hwnd, Int32 msg, IntPtr wParam, IntPtr lParam, Boolean& handled)
   at MS.Win32.HwndWrapper.WndProc(IntPtr hwnd, Int32 msg, IntPtr wParam, IntPtr lParam, Boolean& handled)
   at MS.Win32.HwndSubclass.DispatcherCallbackOperation(Object o)
   at System.Windows.Threading.ExceptionWrapper.InternalRealCall(Delegate callback, Object args, Int32 numArgs)
   at System.Windows.Threading.ExceptionWrapper.TryCatchWhen(Object source, Delegate callback, Object args, Int32 numArgs, Delegate catchHandler)
   at System.Windows.Threading.Dispatcher.LegacyInvokeImpl(DispatcherPriority priority, TimeSpan timeout, Delegate method, Object args, Int32 numArgs)
   at MS.Win32.HwndSubclass.SubclassWndProc(IntPtr hwnd, Int32 msg, IntPtr wParam, IntPtr lParam)
   at MS.Win32.UnsafeNativeMethods.DispatchMessage(MSG& msg)
   at System.Windows.Threading.Dispatcher.PushFrameImpl(DispatcherFrame frame)
   at System.Windows.Threading.Dispatcher.PushFrame(DispatcherFrame frame)
   at System.Windows.Threading.Dispatcher.Run()
   at System.Windows.Application.RunDispatcher(Object ignore)
   at System.Windows.Application.RunInternal(Window window)
   at System.Windows.Application.Run(Window window)
   at Eto.Wpf.Forms.ApplicationHandler.Run()
   at Eto.Forms.Application.Run(Form mainForm)
   at OpenTabletDriver.UX.Wpf.MainClass.Main(String[] args) in C:\Users\Infinity\OpenTabletDriver\OpenTabletDriver.UX.Wpf\Program.cs:line 11

Event Xml:
<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
  <System>
    <Provider Name=".NET Runtime" />
    <EventID Qualifiers="0">1026</EventID>
    <Version>0</Version>
    <Level>2</Level>
    <Task>0</Task>
    <Opcode>0</Opcode>
    <Keywords>0x80000000000000</Keywords>
    <TimeCreated SystemTime="2020-11-23T00:32:23.3692302Z" />
    <EventRecordID>7057</EventRecordID>
    <Correlation />
    <Execution ProcessID="0" ThreadID="0" />
    <Channel>Application</Channel>
    <Computer>DESKTOP-S7THMED</Computer>
    <Security />
  </System>
  <EventData>
    <Data>Application: OpenTabletDriver.UX.Wpf.exe
CoreCLR Version: 5.0.20.51904
.NET Version: 5.0.0
Description: The process was terminated due to an unhandled exception.
Exception Info: StreamJsonRpc.ConnectionLostException: The JSON-RPC connection with the remote party was lost before the request could complete.
[... 8 lines cut ...]
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state)
--- End of stack trace from previous location ---
   at MS.Internal.CulturePreservingExecutionContext.Run(CulturePreservingExecutionContext executionContext, ContextCallback callback, Object state)
   at System.Windows.Threading.DispatcherOperation.Invoke()
   at System.Windows.Threading.Dispatcher.ProcessQueue()
   at System.Windows.Threading.Dispatcher.WndProcHook(IntPtr hwnd, Int32 msg, IntPtr wParam, IntPtr lParam, Boolean&amp; handled)
   at MS.Win32.HwndWrapper.WndProc(IntPtr hwnd, Int32 msg, IntPtr wParam, IntPtr lParam, Boolean&amp; handled)
   at MS.Win32.HwndSubclass.DispatcherCallbackOperation(Object o)
   at System.Windows.Threading.ExceptionWrapper.InternalRealCall(Delegate callback, Object args, Int32 numArgs)
   at System.Windows.Threading.ExceptionWrapper.TryCatchWhen(Object source, Delegate callback, Object args, Int32 numArgs, Delegate catchHandler)
[... cut ...]
   at System.Windows.Threading.Dispatcher.PushFrame(DispatcherFrame frame)
   at System.Windows.Threading.Dispatcher.Run()
   at System.Windows.Application.RunDispatcher(Object ignore)
   at System.Windows.Application.RunInternal(Window window)
   at System.Windows.Application.Run(Window window)
   at Eto.Wpf.Forms.ApplicationHandler.Run()
   at Eto.Forms.Application.Run(Form mainForm)
   at OpenTabletDriver.UX.Wpf.MainClass.Main(String[] args) in C:\Users\Infinity\OpenTabletDriver\OpenTabletDriver.UX.Wpf\Program.cs:line 11
</Data>
  </EventData>
</Event>

Mrcubix, 2020-11-23:
try deleting your setting and starting OTD again real quick

Mrcubix, 2020-11-23:
they are located in %localappdata%/opentabletdriver/

Jahquise, 2020-11-23:
ok it fixed it

Jahquise, 2020-11-23:
thank you for your help

Mrcubix, 2020-11-23:
tho it might happen again if it was working earlier, so i advice you to keep a working version of the settings in a backup folder

Jahquise, 2020-11-23:
ok

closed 2020-11-23 by Jahquise