OpenTabletDriver/OpenTabletDriver #468: Does opentabletdriver support pen pressure?

Turk3ysandwich, 2020-11-23:
# Support Request
 I use the tablet for osu mostly but some drawing and autodesk sketchbook supports pen pressure. Am I doing something wrong?

solonovamax, 2020-11-23:
It supports pen pressure in some applications. I know that it works in krita, but doesn't seem to work for me in gimp.
(You need to switch it to "artist mode" to enable it. This is for linux, idk for windows.)

Mrcubix, 2020-11-23:
there is a plugin just for that, https://github.com/X9VoiD/VoiDPlugins/releases/download/v0.2.1/WindowsInk.zip put that in %localappdata%/opentabletdriver/Plugins

Mrcubix, 2020-11-23:
also you need to install this https://github.com/X9VoiD/vmulti-bin/releases/tag/v1.0

Mrcubix, 2020-11-23:
when done, switch to artist mode in otd

X9VoiD, 2020-11-23:
This issue should be closed as we already have pressure support for Windows in the form of a plugin

InfinityGhost, 2020-11-23:
> This issue should be closed as we already have pressure support for Windows in the form of a plugin

Agreed.

closed 2020-11-23 by InfinityGhost

Turk3ysandwich, 2020-11-23:
> when done, switch to artist mode in otd

It's me spaghetti lol