OpenTabletDriver/OpenTabletDriver #469: OTD breaks when application with admin permissions is open. [Windows]

solonovamax, 2020-11-23:
## Description

When an application with administrator permissions is in focus on windows, OTD can no longer control mouse input.

Nothing is logged, it just loses control.

This applies to the "are you sure you want to give this app administrator permissions" dialogue and normal apps launched with admin perms.
## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System | Windows |
| Software Version | 0.4.2 |
| Tablet           |  Star G640 |

I'm not normally on windows, I'm just using my tablet with another computer and happened to notice this and decided to report it. As such, my ability to give debug information is limited.

InfinityGhost, 2020-11-23:
This is impossible to fix with our current API, it's a Windows "security" feature. It filters any `SendInput` API calls when an elevated process is in the foreground and the only workaround is to use a kernel level driver (like vmulti) to bypass this.

closed 2020-11-23 by InfinityGhost

solonovamax, 2020-11-23:
rip, that's what I thought but wanted to open it anyways.

InfinityGhost, 2020-11-23:
A separate OutputMode plugin could be developed to use a kernel driver like vmulti to resolve issues like this, but internally we're sticking to the native APIs.

Mrcubix, 2020-11-23:
just don't use admin mode, and don't use admin mode on osu either

solonovamax, 2020-11-23:
> just don't use admin mode, and don't use admin mode on osu either

I know that's the obvious solution, but not a complete one. I just opened the issue for completeness.

X9VoiD, 2020-11-23:
There's VMultiMode on VoiDPlugin repo