OpenTabletDriver/OpenTabletDriver #470: Left Click does not work properly in osu! (clicks at different places)

grimalkin-esmeralda, 2020-11-23:
## Description
The left Click works perfectly fine in all other applications, but in osu! it sometimes clicks at completely different locations.
Like, starting a song that is 10 spots down, not being able to quit a map bc instead of hitting the button it always opens up the "option" slider from the bottom, when scrolling through songs (which only sometimes works) it scrolls back to the top once i release the left click. etc.

Really nice Driver, but at this point not a replacement for the Gaomon stock ones sadly?

## System Information:
Win7
v0.4.2
Gaomon S620

Mrcubix, 2020-11-23:
make sure you unbound everything according to https://youtu.be/A6JHLBiRRkQ
also this is a replacement

grimalkin-esmeralda, 2020-12-10:
> 
> 
> make sure you unbound everything (every button) according to https://youtu.be/A6JHLBiRRkQ
> also this is a replacement
> also uc_logic tablet are fucked, gaomon s620 is one of them, and i ripped the button off it cause if you push the button, it will stop working with any driver

i did everything correctly.
it just doesnt work, and a proper answer would be appreciated, what is "invalid" supposed to mean?

grimalkin-esmeralda, 2020-12-10:
Ok, coincidentally i just found out what exactly happens:
whenever i press the tip onto the tablet, the cursor jumps to the lower right corner of the osu! window for a split second and clicks there and fucks everything up.

what could be the reason for that?

InfinityGhost, 2020-12-10:
> Ok, coincidentally i just found out what exactly happens:
> whenever i press the tip onto the tablet, the cursor jumps to the lower right corner of the osu! window for a split second and clicks there and fucks everything up.
> 
> what could be the reason for that?

Could be ingame sensitivity settings or raw input

X9VoiD, 2021-01-22:
Obsolete issue, so unless the issue can be reproduced by other Gaomon S620 users this issue can be closed.

InfinityGhost, 2021-04-06:
> Obsolete issue, so unless the issue can be reproduced by other Gaomon S620 users this issue can be closed.

closed 2021-04-06 by InfinityGhost