OpenTabletDriver/OpenTabletDriver #477: Add support for Huion H320M

rainbowsix6, 2020-11-26:
# Tablet Information

| Name            | Value |
| --------------- | ----- |
| Manufacturer    | HUION |
| Product Name    | Huion Tablet_H320M |
| Width           | 229mm |
| Height          | 143mm |
| Pressure Levels | 8192 |
| LPI             | 5080 |

## Diagnostic Information

[w.txt](https://github.com/InfinityGhost/OpenTabletDriver/files/5602966/w.txt)

X9VoiD, 2020-11-26:
Progress is being made to simplify further adding support to new tablets, so we will come back to this after HidSharpCore's [#14](https://github.com/InfinityGhost/HIDSharpCore/pull/14) comes to OTD

X9VoiD, 2021-01-22:
@rainbowsix6 Use [0.5.0](https://github.com/InfinityGhost/OpenTabletDriver/releases/tag/v0.5.0) release and do OTD -> Help -> Export Diagnostics so we may take advantage of OTD's new capability to know which endpoints are available to open. Upload the generated file here.

kourm, 2021-04-16:
Here is a (somewhat redacted) log for the 320m
[otd.diag.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6326328/otd.diag.txt)
and the relevant output of `sudo lsusb -vvv` 
[320_lsusb.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6326341/320_lsusb.txt)
as well as the two report descriptors (identified as a hid device):
`hexdump /sys/bus/usb/drivers/usbhid/4-1.3.2\:1.0/0003\:256C\:006D.0010/report_descriptor` 
[320_desc_in0.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6326357/320_desc_in0.txt)
and (identified as a hid keyboard)
`hexdump /sys/bus/usb/drivers/usbhid/4-1.3.2\:1.1/0003\:256C\:006D.0011/report_descriptor`
[320_desc_in1.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6326359/320_desc_in1.txt)

Hope that's sufficient, otherwise I can try to do more

X9VoiD, 2022-01-16:
Stale duplicate of #1927

closed 2022-01-16 by X9VoiD