OpenTabletDriver/OpenTabletDriver #479: Bind OpenTabletDriver menu to button

Fidoo001, 2020-11-26:
Currently there is no way to open the driver menu without clicking the icon on task bar. Please add option to bind it to keyboard button and tablet express keys. Thanks

InfinityGhost, 2020-11-26:
So what you want is a binding that will show the OpenTabletDriver GUI?

Fidoo001, 2020-11-26:
Yes, I meant the OpenTabletDriver GUI

Aksumxd, 2020-11-27:
idk if this belongs here, but I personally made a .bat file that opens osu!, my drivers, and REAL in one click
```
start C:\OpenTabletDriver\OpenTabletDriver.Installer.exe
start C:\Users\Akseli\Downloads\REAL.exe
start C:\Users\Akseli\AppData\Local\osu!\osu!.exe
```
Maybe you'll find it useful

Mrcubix, 2020-11-27:
that doesn't work
since the installer and the portable start a new instance of OTD instead of re-using the active one

X9VoiD, 2021-01-07:
This issue may be resolved by wrapping the `EventWaitHandle` logic from #609 to an `IBinding` and call `EventWaitHandle.Set()`.

AbstractQbit, 2021-01-07:
`EventWaitHandle` is not portable, it works correctly only on Windows. As [Infinity mentioned](https://discord.com/channels/615607687467761684/628651971267788800/794405589362933790), it's better to handle this stuff through existing rpc between daemon and ux instances.

George-lewis, 2021-02-25:
Is this resolved? I think AutoHotKey + Singleton UI pattern enable this usecase

Mrcubix, 2021-02-25:
> Is this resolved? I think AutoHotKey + Singleton UI pattern enable this usecase

OP is asking for a new type of binding for UX function such as A function that take the UX in argument and run Show() like used in the Tray icon, but as of 0.5.1, a workaround is just using ahk and this script in OTD using script runner.

```ahk
SetTitleMatchMode, 2

DetectHiddenWindows, On

IfWinExist OpenTabletDriver

WinActivate
```

However, this is only a workaround, a proper implementation in OTD is still needed but not a priority right now, so if you want to try implementing something yourself, feel free to do so

George-lewis, 2021-02-25:
Oh I see, I'll consider looking into it.

InfinityGhost, 2021-02-25:
> Oh I see, I'll consider looking into it.

#635 needs to be finished for this to happen. We want it to be platform agnostic, and the best way to do that is directly through the existing RPC since bindings are executed on the daemon, and not the UI.

George-lewis, 2021-02-25:
I'll leave this alone for now

X9VoiD, 2021-02-25:
#635 also already adds the binding.