OpenTabletDriver/OpenTabletDriver #484: Informative hoverboxes for filter options

Aksumxd, 2020-11-27:
One of the few things I missed after switching over from the Hawku drivers.

[image]

Of course not really that important, but it would help less tech-savvy people understand them.

Mrcubix, 2020-11-27:
Basically, you're asking to add tooltips to a dozen plugin/tools, so adding around 30+ tools tips.
for indications about what does what.
MeL has a wiki that can be used for these: https://github.com/X9VoiD/VoiDPlugins/wiki/MeL
as for the other plugins, i don't know if the contributors will want to work on implementing those tooltips in their filters/tools.

Aksumxd, 2020-11-27:
This was mostly meant for the official filters, mb
maybe move to https://github.com/InfinityGhost/TabletDriverFilters/issues

InfinityGhost, 2020-11-28:
The formatting wouldn't be as pretty, but yes this does belong more on TabletDriverFilters. Will transfer it there.