OpenTabletDriver/OpenTabletDriver #485: OpenTabletDriver.UX.Wpf.exe not starting

jupiteria47, 2020-11-28:
## Description

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System | Microsoft Windows
| Software Version | Windows 7 64-bit SP1
| Tablet           | Huion H430P

InfinityGhost, 2020-11-28:
Software version is asking for OpenTabletDriver version, probably should clarify that in the template lol

Anyway, are you familiar with how to use event viewer?

jupiteria47, 2020-11-28:
idk how to use that

jupiteria47, 2020-11-28:
> Software version is asking for OpenTabletDriver version, probably should clarify that in the template lol
> 
> Anyway, are you familiar with how to use event viewer?

idk what is that and how to use it

Mrcubix, 2020-11-28:
make sure you installed .net 5 Runtime x64

jupiteria47, 2020-11-29:
> make sure you installed .net 5 Runtime x64
> https://dotnet.microsoft.com/download/dotnet/thank-you/runtime-desktop-5.0.0-windows-x64-installer

oh yeah i4got i only have .net framework 4.7

jupiteria47, 2020-11-29:
Update: i have installed .net 5 runtime, nothing still happens when i open the installer

X9VoiD, 2020-11-29:
You may just download `OpenTabletDriver.win-x64.zip` directly from Assets of [latest releases](https://github.com/InfinityGhost/OpenTabletDriver/releases/latest).

jupiteria47, 2020-11-29:
> You may just download `OpenTabletDriver.win-x64.zip` directly from Assets of [latest releases](https://github.com/InfinityGhost/OpenTabletDriver/releases/latest).

this is what i downloaded yesterday and OpenTabletDriver.UX.Wpf.exe doesn't open

jupiteria47, 2020-12-03:
nvm imma just close this issue and download the installer instead of the portable

closed 2020-12-03 by jupiteria47