OpenTabletDriver/OpenTabletDriver #507: Add support for Deco mini7

bleckers, 2020-12-03:
# Tablet Information

| Name            | Value |
| --------------- | ----- |
| Manufacturer    | XP-PEN
| Product Name    | Deco mini7
| Width           | 177.8
| Height          | 111.1
| Pressure Levels | 8192
| LPI             | 5080

Tested with this config and things seem to work well:

```
{
  "Name": "XP-Pen Deco mini7",
  "DigitizerIdentifiers": [
    {
      "Width": 177.8,
      "Height": 111.1,
      "MaxX": 35560.0,
      "MaxY": 22220.0,
      "MaxPressure": 8191,
      "ActiveReportID": {
        "Start": 128,
        "StartInclusive": false,
        "End": null,
        "EndInclusive": false
      },
      "VendorID": 10429,
      "ProductID": 2344,
      "InputReportLength": 12,
      "OutputReportLength": 12,
      "ReportParser": null,
      "FeatureInitReport": null,
      "OutputInitReport": "ArAE",
      "DeviceStrings": {},
      "InitializationStrings": []
    }
  ],
  "AuxilaryDeviceIdentifiers": [],
  "Attributes": {
    "libinputoverride": "1"
  }
}

```

Mrcubix, 2020-12-03:
a diagnostic file i required to make any config, to generate one, open the UX, go to help > export diagnostic, give it a name and save it somewhere then upload it here

InfinityGhost, 2020-12-04:
> ``` "OutputInitReport": "ArAE",```

This is invalid Base64, I don't know how you managed to get this to work.

bleckers, 2020-12-04:
That was taken from the Deco 01 v2 JSON file. I just changed vendorID, productID, width, height, maxX, maxY and reportLenths.

bleckers, 2020-12-04:
Also as for the name change, the manufacturer calls it the "Deco mini7" - https://www.xp-pen.com/product/613.html

InfinityGhost, 2020-12-04:
> Also as for the name change, the manufacturer calls it the "Deco mini7" - https://www.xp-pen.com/product/613.html

Odd naming scheme, will keep like you had it.

AronMS834, 2021-01-31:
In version 0.5.1 I tried my mini7 tablet and the noise reduction and smoothing functions were turned on by default, unlike ctl672 which reads data directly without noise reduction and smoothing.

Mrcubix, 2021-01-31:
Welcome to the Huion world where every huion tablet has hardware smoothing included

AronMS834, 2021-02-01:
> Welcome to the Huion world where every tablet produced by huion has hardware smoothing included
> (XP-Pen, Gaomon, Veikk, UGEE, Huion, ACPEE, ....)
> 
> you can use the reconstructor plugin and its associated wiki to remove that

Ok, I originally thought it was their drivers, so is it just wacom without hardware smoothing? Incidentally, xppen and UGEE are products of the same company. I've asked their after sales support before and they said it was no hardware smoothing. Don't understand what's going on, but I'm switching back to wacom.

AronMS834, 2021-02-01:
> Welcome to the Huion world where every tablet produced by huion has hardware smoothing included
> (XP-Pen, Gaomon, Veikk, UGEE, Huion, ACPEE, ....)
> 
> you can use the reconstructor plugin and its associated wiki to remove that

By the way, thanks for the reply.

X9VoiD, 2021-02-01:
If they don't have any smoothing then reconstructor shouldn't have worked at all since the whole premise of reconstructor is to reverse the applied smoothing algorithm, and if they really don't have hardware smoothing then their input latency should've been lower than wacom due to higher "advertised" polling rate.

bespokecomp, 2021-08-19:
> 
> # Tablet Information
> Name 	Value
> Manufacturer 	XP-Pen
> Product Name 	Deco mini7
> Width 	177.8
> Height 	111.1
> Pressure Levels 	8192
> LPI 	5080
> 
> Tested with this config and things seem to work well:
> 
> ```
> {
>   "Name": "XP-Pen Deco mini7",
>   "DigitizerIdentifiers": [
>     {
>       "Width": 177.8,
>       "Height": 111.1,
>       "MaxX": 35560.0,
>       "MaxY": 22220.0,
>       "MaxPressure": 8191,
>       "ActiveReportID": {
>         "Start": 128,
>         "StartInclusive": false,
>         "End": null,
>         "EndInclusive": false
>       },
>       "VendorID": 10429,
>       "ProductID": 2344,
>       "InputReportLength": 12,
>       "OutputReportLength": 12,
>       "ReportParser": null,
>       "FeatureInitReport": null,
>       "OutputInitReport": "ArAE",
>       "DeviceStrings": {},
>       "InitializationStrings": []
>     }
>   ],
>   "AuxilaryDeviceIdentifiers": [],
>   "Attributes": {
>     "libinputoverride": "1"
>   }
> }
> ```
> 
> While it is similar to the Deco 01 v2, it doesn't seem to work until In/OutputReportLength is set to 12.

Hi, I have the XP Pen deco mini7.  I can't get the Auxiliary buttons to work.  The config file has InputReportLength set to 12 and the OutputReportLength set to 10.  i tried changing OuputReportLength to 12 like you said.  No change.  
Any ideas?

X9VoiD, 2021-08-19:
This is an old issue that is already resolved in `master` branch, same with your aux button issue.

closed 2021-08-19 by X9VoiD

bespokecomp, 2021-08-19:
> 
> 
> This is an old issue that is already resolved in `master` branch, same with your aux button issue.

Sorry, I don't know how to find the resolution.  I searched through every issue.
You'd really be helping me out if you could point me towards the resolution.

Many thanks

X9VoiD, 2021-08-19:
[Dev builds](https://github.com/OpenTabletDriver/OpenTabletDriver/actions?query=event%3Apush+is%3Acompleted)

bespokecomp, 2021-08-20:
> Use a [dev build](https://github.com/OpenTabletDriver/OpenTabletDriver/actions?query=event%3Apush+is%3Acompleted).

I cannot find any information on here about how one goes about doing this. Please can you point me in the direction of this?
I'd sincerely appreciate it.

jamesbt365, 2021-08-20:
https://github.com/OpenTabletDriver/OpenTabletDriver/actions?query=event%3Apush+is%3Acompleted

Btw you have to be logged in to see them, though you probably will be but heads up

bespokecomp, 2021-08-22:
> 
> 
> https://github.com/OpenTabletDriver/OpenTabletDriver/actions?query=event%3Apush+is%3Acompleted
> 
> Btw you have to be logged in to see them, though you probably will be but heads up

Thanks for replying.  In a nutshell, what is it that I must do to get it to work.  I found this at the link you sent me, but I haven't got any idea what I am looking at or supposed to be looking for?
https://github.com/OpenTabletDriver/OpenTabletDriver/actions/runs/523574624