OpenTabletDriver/OpenTabletDriver #508: Cannot find plugin directory

bleckers, 2020-12-03:
## Description
It seems that if there's a space in the install directory under Windows (e.g. C:\Users\Firstname Lastname\AppData\Local\OpenTabletDriver\bin), it cannot open plugins. "Browse>Open plugins directory" just gives the error: "Windows cannot find 'C:\Users\Firstname'...."

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System | Windows 10
| Software Version | 0.4.2
| Tablet           | Any

InfinityGhost, 2020-12-04:
@X9VoiD

closed 2020-12-05 by InfinityGhost

Pancake-Gang, 2021-01-16:
I don't know if im just dumb, but I still have this problem. I've re-installed it, and yet it still says that I can't find the plugin directory.
It is probally just me, and there is an easy fix to this prolem.

AbstractQbit, 2021-01-16:
@Pancake-Gang if you are using 0.4.2, then the fix won't be there. Use [action builds](https://github.com/InfinityGhost/OpenTabletDriver/actions?query=event%3Apush) or a pre-release from our discord server.

Pancake-Gang, 2021-01-16:
> 
> 
> @Pancake-Gang if you are using 0.4.2, then the fix won't be there. Use [action builds](https://github.com/InfinityGhost/OpenTabletDriver/actions?query=event%3Apush) or a pre-release from our discord server.

Thank you!

PR #513 (2020-12-05): Fix opening folders with spaces on Windows, again
Reuse `SystemInterop.Open()` where possible

## Issues

Fixes #508