OpenTabletDriver/OpenTabletDriver #514: Add support for

KoleckOLP, 2020-12-04:
# Tablet Information

| Name            | Value |
| --------------- | ----- |
| Manufacturer    | Genius
| Product Name    | WizardPen 5x4
| Width           | 5.5 in
| Height          | 3.9 in
| Pressure Levels | 512 or 1024
| LPI             | 4064

I'm sorry I couldn't confirm these it's hard to find info about it.

## Diagnostic Information

[Diagnostic.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5645620/Diagnostic.zip)

InfinityGhost, 2020-12-05:
Try this build: https://github.com/InfinityGhost/OpenTabletDriver/actions/runs/401746432

If it does work, we still need to verify some data so that dimensions are correct.
1. Open the Tablet Debugger
2. Move your pen to the bottom right corner of your tablet, where X & Y on the tablet parser is at its highest value.
3. Screenshot and send here

If it doesn't work, I'll link another version later for you to try.

KoleckOLP, 2020-12-05:
```
[Detect:Info]   Searching for tablet 'Gaomon 1060 Pro'
[Detect:Info]   Searching for tablet 'Gaomon M106K'
[Detect:Info]   Searching for tablet 'Gaomon S56K'
[Detect:Info]   Searching for tablet 'Gaomon S620'
[Detect:Info]   Searching for tablet 'Genius WizardPen 5x4'
[Detect:Warning]        More than 1 matching digitizer has been found.
[Detect:Debug]  Using device 'TABLET DEVICE'.
[Detect:Debug]  Using report parser type 'OpenTabletDriver.Tablet.TabletReportParser'.
[Detect:Debug]  Device path: \\?\hid#vid_5543&pid_0004&col03#6&7474765&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}
[IOException:Error]     Failed to open stream
   at OpenTabletDriver.Devices.DeviceReader`1..ctor(HidDevice device, IReportParser`1 reportParser) in D:\a\OpenTabletDriver\OpenTabletDriver\OpenTabletDriver\Devices\DeviceReader.cs:line 30
   at OpenTabletDriver.Driver.InitializeDigitizerDevice(HidDevice tabletDevice, DigitizerIdentifier tablet, IReportParser`1 reportParser) in D:\a\OpenTabletDriver\OpenTabletDriver\OpenTabletDriver\Driver.cs:line 163
   at OpenTabletDriver.Driver.TryMatchDigitizer(TabletConfiguration config, DigitizerIdentifier& digitizerIdentifier) in D:\a\OpenTabletDriver\OpenTabletDriver\OpenTabletDriver\Driver.cs:line 105
[Detect:Debug]  Using device 'TABLET DEVICE'.
[Detect:Debug]  Using report parser type 'OpenTabletDriver.Tablet.TabletReportParser'.
[Detect:Debug]  Device path: \\?\hid#vid_5543&pid_0004&col01#6&7474765&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
[Detect:Info]   Found tablet 'Genius WizardPen 5x4'
[Settings:Info] Output mode: Absolute Mode
[Settings:Info] Display area: [1920x1080@<960  540>:0°],
[Settings:Info] Tablet area: [139,7x99,06@<69,85  49,53>:0°],
[Settings:Info] Clipping: Enabled
[Settings:Info] Ignoring reports outside area: Disabled
[Settings:Info] Tip Binding: [Mouse Button Binding: Left]@1%
[Settings:Info] Pen Bindings: [0, ], [1, ]
[Settings:Info] Express Key Bindings: [0, ], [1, ], [2, ], [3, ], [4, ], [5, ]
[Settings:Info] Driver is auto-enabled.
[IPC:Debug]     Connected to a client.
[Display:Debug] DPI Awareness enabled
[Settings:Info] Output mode: Absolute Mode
[Settings:Info] Display area: [1920x1080@<960  540>:0°],
[Settings:Info] Tablet area: [139,7x99,06@<69,85  49,53>:0°],
[Settings:Info] Clipping: Enabled
[Settings:Info] Ignoring reports outside area: Disabled
[Settings:Info] Tip Binding: [Mouse Button Binding: Left]@1%
[Settings:Info] Pen Bindings: [0, ], [1, ]
[Settings:Info] Express Key Bindings: [0, ], [1, ], [2, ], [3, ], [4, ], [5, ]
[Settings:Info] Driver is auto-enabled.
[Settings:Info] Output mode: Absolute Mode
[Settings:Info] Display area: [1920x1080@<960  540>:0°],
[Settings:Info] Tablet area: [139,7x99,06@<69,85  49,53>:0°],
[Settings:Info] Clipping: Enabled
[Settings:Info] Ignoring reports outside area: Disabled
[Settings:Info] Tip Binding: [Mouse Button Binding: Left]@1%
[Settings:Info] Pen Bindings: [0, ], [1, ]
[Settings:Info] Express Key Bindings: [0, ], [1, ], [2, ], [3, ], [4, ], [5, ]
[Settings:Info] Driver is auto-enabled.
[Detect:Info]   Searching for tablet 'Gaomon 1060 Pro'
[Detect:Info]   Searching for tablet 'Gaomon M106K'
[Detect:Info]   Searching for tablet 'Gaomon S56K'
[Detect:Info]   Searching for tablet 'Gaomon S620'
[Detect:Info]   Searching for tablet 'Genius WizardPen 5x4'
[Detect:Warning]        More than 1 matching digitizer has been found.
[Detect:Debug]  Using device 'TABLET DEVICE'.
[Device:Debug]  A device stream was disposed.
[Detect:Debug]  Using report parser type 'OpenTabletDriver.Tablet.TabletReportParser'.
[Detect:Debug]  Device path: \\?\hid#vid_5543&pid_0004&col03#6&7474765&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}
[IOException:Error]     Failed to open stream
   at OpenTabletDriver.Devices.DeviceReader`1..ctor(HidDevice device, IReportParser`1 reportParser) in D:\a\OpenTabletDriver\OpenTabletDriver\OpenTabletDriver\Devices\DeviceReader.cs:line 30
   at OpenTabletDriver.Driver.InitializeDigitizerDevice(HidDevice tabletDevice, DigitizerIdentifier tablet, IReportParser`1 reportParser) in D:\a\OpenTabletDriver\OpenTabletDriver\OpenTabletDriver\Driver.cs:line 163
   at OpenTabletDriver.Driver.TryMatchDigitizer(TabletConfiguration config, DigitizerIdentifier& digitizerIdentifier) in D:\a\OpenTabletDriver\OpenTabletDriver\OpenTabletDriver\Driver.cs:line 105
[Detect:Debug]  Using device 'TABLET DEVICE'.
[Detect:Debug]  Using report parser type 'OpenTabletDriver.Tablet.TabletReportParser'.
[Detect:Debug]  Device path: \\?\hid#vid_5543&pid_0004&col01#6&7474765&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
[Detect:Info]   Found tablet 'Genius WizardPen 5x4'
```

The tablet seems to get detected, but changing area or anything does not work. And the tablet debugger stays empty even if I hover or tap the tablet.

InfinityGhost, 2020-12-07:
It's detected, but it failed to start reading device data. The only other configuration we can give a shot at right now is this build I'm linking, and it's likely a step in the wrong direction.
https://github.com/InfinityGhost/OpenTabletDriver/actions/runs/402977157

KoleckOLP, 2020-12-07:
```
[Display:Debug] DPI Awareness enabled
[Settings:Info] Output mode: Absolute Mode
[Settings:Info] Display area: [3840x1080@<1920  540>:0°],
[Settings:Info] Tablet area: [0x0@<0  0>:0°],
[Settings:Info] Clipping: Enabled
[Settings:Info] Ignoring reports outside area: Disabled
[Settings:Info] Tip Binding: [Mouse Button Binding: Left]@1%
[Settings:Info] Pen Bindings: [0, ], [1, ]
[Settings:Info] Express Key Bindings: [0, ], [1, ], [2, ], [3, ], [4, ], [5, ]
[Settings:Info] Driver is auto-enabled.
[IPC:Debug]     Connected to a client.
[Display:Debug] DPI Awareness enabled
[Detect:Info]   Searching for tablet 'Gaomon 1060 Pro'
[Detect:Info]   Searching for tablet 'Gaomon M106K'
[Detect:Info]   Searching for tablet 'Gaomon S56K'
[Detect:Info]   Searching for tablet 'Gaomon S620'
[Detect:Info]   Searching for tablet 'Genius WizardPen 5x4'
[Detect:Debug]  Using device 'TABLET DEVICE'.
[Detect:Debug]  Using report parser type 'OpenTabletDriver.Tablet.TabletReportParser'.
[Detect:Debug]  Device path: \\?\hid#vid_5543&pid_0004&col02#6&7474765&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}
[IOException:Error]     Failed to open stream
   at OpenTabletDriver.Devices.DeviceReader`1..ctor(HidDevice device, IReportParser`1 reportParser) in D:\a\OpenTabletDriver\OpenTabletDriver\OpenTabletDriver\Devices\DeviceReader.cs:line 30
   at OpenTabletDriver.Driver.InitializeDigitizerDevice(HidDevice tabletDevice, DigitizerIdentifier tablet, IReportParser`1 reportParser) in D:\a\OpenTabletDriver\OpenTabletDriver\OpenTabletDriver\Driver.cs:line 163
   at OpenTabletDriver.Driver.TryMatchDigitizer(TabletConfiguration config, DigitizerIdentifier& digitizerIdentifier) in D:\a\OpenTabletDriver\OpenTabletDriver\OpenTabletDriver\Driver.cs:line 105
[Detect:Info]   Searching for tablet 'Huion GT-133'
[Detect:Info]   Searching for tablet 'Huion H430P'
[Detect:Info]   Searching for tablet 'Huion H610 Pro V2'
[Detect:Info]   Searching for tablet 'Huion H610 Pro'
[Detect:Info]   Searching for tablet 'Huion H640P'
[Detect:Info]   Searching for tablet 'Huion HS610'
[Detect:Info]   Searching for tablet 'Huion HS611'
[Detect:Info]   Searching for tablet 'Huion HS64'
[Detect:Info]   Searching for tablet 'Huion Inspiroy Q11K'
[Detect:Info]   Searching for tablet 'Huion New 1060 Plus'
[Detect:Info]   Searching for tablet 'Huion osu!tablet'
[Detect:Info]   Searching for tablet 'Huion Q620M'
[Detect:Info]   Searching for tablet 'VEIKK A15 Pro'
[Detect:Info]   Searching for tablet 'VEIKK A15'
[Detect:Info]   Searching for tablet 'VEIKK A30'
[Detect:Info]   Searching for tablet 'VEIKK A50'
[Detect:Info]   Searching for tablet 'VEIKK S640'
[Detect:Info]   Searching for tablet 'Wacom CTE-430'
[Detect:Info]   Searching for tablet 'Wacom CTE-440'
[Detect:Info]   Searching for tablet 'Wacom CTE-460'
[Detect:Info]   Searching for tablet 'Wacom CTH-470'
[Detect:Info]   Searching for tablet 'Wacom CTH-480'
[Detect:Info]   Searching for tablet 'Wacom CTH-490'
[Detect:Info]   Searching for tablet 'Wacom CTH-670'
[Detect:Info]   Searching for tablet 'Wacom CTH-680'
[Detect:Info]   Searching for tablet 'Wacom CTH-690'
[Detect:Info]   Searching for tablet 'Wacom CTL-4100 Bluetooth'
[Detect:Info]   Searching for tablet 'Wacom CTL-4100'
[Detect:Info]   Searching for tablet 'Wacom CTL-4100WL'
[Detect:Info]   Searching for tablet 'Wacom CTL-460'
[Detect:Info]   Searching for tablet 'Wacom CTL-470'
[Detect:Info]   Searching for tablet 'Wacom CTL-471'
[Detect:Info]   Searching for tablet 'Wacom CTL-472'
[Detect:Info]   Searching for tablet 'Wacom CTL-480'
[Detect:Info]   Searching for tablet 'Wacom CTL-490'
[Detect:Info]   Searching for tablet 'Wacom CTL-6100'
[Detect:Info]   Searching for tablet 'Wacom CTL-671'
[Detect:Info]   Searching for tablet 'Wacom CTL-672'
[Detect:Info]   Searching for tablet 'Wacom CTL-680'
[Detect:Info]   Searching for tablet 'Wacom CTL-690'
[Detect:Info]   Searching for tablet 'Wacom FT-0405-U'
[Detect:Info]   Searching for tablet 'Wacom MTE-450'
[Detect:Info]   Searching for tablet 'Wacom PTH-451'
[Detect:Info]   Searching for tablet 'Wacom PTH-651'
[Detect:Info]   Searching for tablet 'Wacom PTH-660 Bluetooth'
[Detect:Info]   Searching for tablet 'Wacom PTH-660'
[Detect:Info]   Searching for tablet 'Wacom PTH-850'
[Detect:Info]   Searching for tablet 'Wacom PTK-440'
[Detect:Info]   Searching for tablet 'XP-Pen Deco 01 v2'
[Detect:Info]   Searching for tablet 'XP-Pen Deco 01'
[Detect:Info]   Searching for tablet 'XP-Pen Deco 02'
[Detect:Info]   Searching for tablet 'XP-Pen Deco 03'
[Detect:Info]   Searching for tablet 'XP-Pen G430'
[Detect:Info]   Searching for tablet 'XP-Pen G430S'
[Detect:Info]   Searching for tablet 'XP-Pen G540 Pro'
[Detect:Info]   Searching for tablet 'XP-Pen G540'
[Detect:Info]   Searching for tablet 'XP-Pen G640 v2'
[Detect:Info]   Searching for tablet 'XP-Pen G640'
[Detect:Info]   Searching for tablet 'XP-Pen G640s'
[Detect:Info]   Searching for tablet 'XP-Pen G960 v2'
[Detect:Info]   Searching for tablet 'XP-Pen Star 03'
[Detect:Info]   No tablet found.
[Settings:Info] Output mode: Absolute Mode
[Settings:Info] Display area: [3840x1080@<1920  540>:0°],
[Settings:Info] Tablet area: [0x0@<0  0>:0°],
[Settings:Info] Clipping: Enabled
[Settings:Info] Ignoring reports outside area: Disabled
[Settings:Info] Tip Binding: [Mouse Button Binding: Left]@1%
[Settings:Info] Pen Bindings: [0, ], [1, ]
[Settings:Info] Express Key Bindings: [0, ], [1, ], [2, ], [3, ], [4, ], [5, ]
[Settings:Info] Driver is auto-enabled.
[IPC:Debug]     Disconnected from a client.
```
With this one it does not even get detected, so you were correct

X9VoiD, 2021-01-22:
@KoleckOLP use [0.5.0](https://github.com/InfinityGhost/OpenTabletDriver/releases/tag/v0.5.0) and export another diagnostics using that release.

KoleckOLP, 2021-01-22:
`[Detect:Info]	Searching for tablet 'Gaomon 1060 Pro'
[Detect:Info]	Searching for tablet 'Gaomon M106K'
[Detect:Info]	Searching for tablet 'Gaomon S56K'
[Detect:Info]	Searching for tablet 'Gaomon S620'
[Detect:Info]	Searching for tablet 'Genius WizardPen 5x4'
[Detect:Info]	Found tablet 'Genius WizardPen 5x4'
[Settings:Info]	Output mode: Absolute Mode
[Settings:Info]	Display area: [1920x1080@<960  540>:0°],
[Settings:Info]	Tablet area: [45x25,3125@<32  22>:0°],
[Settings:Info]	Clipping: Enabled
[Settings:Info]	Ignoring reports outside area: Disabled
[Settings:Info]	Tip Binding: [Mouse Button Binding: Left]@25%
[Settings:Info]	Pen Bindings: [0, ], [1, ]
[Settings:Info]	Express Key Bindings: [0, ], [1, ], [2, ], [3, ], [4, ], [5, ]
[Settings:Info]	Driver is auto-enabled.
[Settings:Info]	Output mode: Absolute Mode
[Settings:Info]	Display area: [1920x1080@<960  540>:0°],
[Settings:Info]	Tablet area: [45x25,3125@<32  22>:0°],
[Settings:Info]	Clipping: Enabled
[Settings:Info]	Ignoring reports outside area: Disabled
[Settings:Info]	Tip Binding: [Mouse Button Binding: Left]@25%
[Settings:Info]	Pen Bindings: [0, ], [1, ]
[Settings:Info]	Express Key Bindings: [0, ], [1, ], [2, ], [3, ], [4, ], [5, ]
[Settings:Info]	Driver is auto-enabled.
[Settings:Info]	Output mode: Absolute Mode
[Settings:Info]	Display area: [1920x1080@<960  540>:0°],
[Settings:Info]	Tablet area: [45x25,3125@<32  22>:0°],
[Settings:Info]	Clipping: Enabled
[Settings:Info]	Ignoring reports outside area: Disabled
[Settings:Info]	Tip Binding: [Mouse Button Binding: Left]@25%
[Settings:Info]	Pen Bindings: [0, ], [1, ]
[Settings:Info]	Express Key Bindings: [0, ], [1, ], [2, ], [3, ], [4, ], [5, ]
[Settings:Info]	Driver is auto-enabled.
[Detect:Info]	Searching for tablet 'Gaomon 1060 Pro'
[Detect:Info]	Searching for tablet 'Gaomon M106K'
[Detect:Info]	Searching for tablet 'Gaomon S56K'
[Detect:Info]	Searching for tablet 'Gaomon S620'
[Detect:Info]	Searching for tablet 'Genius WizardPen 5x4'
[Detect:Info]	Found tablet 'Genius WizardPen 5x4'
`
[image]

If I use the config with input length 8 the tablet get's detected,
but it shows nothing in the tablet debugger.
With the length of 6 the tablet will not get detected, I do not know what else can I try.

KoleckOLP, 2022-02-13:
Is there a posibility of revisiting this?

gonX, 2022-11-07:
Yes, please upload new full diagnostics from the latest release. Sorry for the delay in getting your issue covered, there are a lot of them.

KoleckOLP, 2022-11-08:
Sorry for late reply i will send you the diagnostic later today.

KoleckOLP, 2022-11-08:
[GeniusWizardPen5x4.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/9962539/GeniusWizardPen5x4.zip)\

Here is diagnostic, the tablet does not get detected because I don't have the config files anymore and can't download them because it says that the builds have expired.

gonX, 2025-11-10:
Sorry for not getting back to this sooner. This issue ended up in a milestone that ended up being very far into the future.

Looking at the information provided, it seems that the endpoint with `OutputReportLength == 8` does not report tablet data. This likely means that the device needs WinUSB on a particular interface.

Newer versions of OpenTabletDriver include interface details, so we'll need a tablet owner to provide a fresh diagnostics file. Sorry for the inconvenience.