OpenTabletDriver/OpenTabletDriver #516: Add support for

Spoiler135, 2020-12-05:
# Tablet Information

| Name            | Value |
| --------------- | ----- |
| Manufacturer    | Xp-Pen |
| Product Name    | Star G540 |
| Width           | 140mm |
| Height          | 120mm |
| Pressure Levels | 2048 |
| LPI             | 5080 |

## Diagnostic Information

[w.txt](https://github.com/InfinityGhost/OpenTabletDriver/files/5646905/w.txt)

X9VoiD, 2020-12-05:
Close OTD, extract [G540.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5646970/G540.zip) to `Configuration\XP-Pen\G540.json`

Spoiler135, 2020-12-05:
I'm sorry but I can't go to the right side of monitor
[image]

X9VoiD, 2020-12-05:
OTD -> Tablet -> Tablet Debugger

Hover around the edges and try to get the highest possible value for Position

Spoiler135, 2020-12-05:
I got Position:[45707, 29208]

X9VoiD, 2020-12-05:
Make sure that *that* is the highest possible value since even just a single digit error will cause issues

Spoiler135, 2020-12-05:
Sorry, it turned out to be Position:[45720, 29210]

X9VoiD, 2020-12-05:
[G540.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5647709/G540.zip)

Spoiler135, 2020-12-05:
It works fine. Thank you so much!

InfinityGhost, 2020-12-05:
Reopening until the configuration is merged to master.

closed 2020-12-05 by InfinityGhost