OpenTabletDriver/OpenTabletDriver #517: ArgumentOutOfRangeException when detecting tablets

zsinx6, 2020-12-05:
## Description
I'm having issues with the Gaomon S620:
`[ArgumentOutOfRangeException:Error]	Value range is [0, 15]. (Parameter 'value')`
Resulting in the tablet not being detected

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System | Arch Linux 
| Software Version | Master build from AUR
| Tablet           | Gaomon S620

X9VoiD, 2020-12-05:
Check wiki on how to blacklist `hid_uclogic` and reload udev

zsinx6, 2020-12-05:
Solved, thank you!

closed 2020-12-05 by zsinx6

gonX, 2020-12-05:
I've seen a few people with this issue now, and the fix has always been to check if any kernel modules that can grab the device are still loaded. Perhaps the [Wiki entry](https://github.com/InfinityGhost/OpenTabletDriver/wiki/Linux-FAQ#argumentoutofrangeexception-value-0-15) should be changed so it doesn't just ask people to install the udev rules?
> Build and install the udev rules with the script above.

I'd be willing to submit a PR if there is interest.

InfinityGhost, 2020-12-05:
> I've seen a few people with this issue now, and the fix has always been to check if any kernel modules that can grab the device are still loaded. Perhaps the [Wiki [...]

If there's a good way to check what module is controlling a device this could be automated in a script quite easily.