OpenTabletDriver/OpenTabletDriver #524: Add support for Wacom DTK-1300 (Cintiq 13HD)

Joromonni, 2020-12-08:
Don't know if it's even possible to work with this thing but here's the details

# Tablet Information

| Name            | Value |
| --------------- | ----- |
| Manufacturer    |Wacom
| Product Name    |DTK-1300
| Width           | 375mm
| Height          |248mm
| Pressure Levels |2048
| LPI             | 5080

## Diagnostic Information

[Diagnostics.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5661016/Diagnostics.zip)

Mrcubix, 2020-12-08:
[Uploading DTK-1300.zip…]()

Joromonni, 2020-12-08:
I dragged the config to the configurations > wacom - folder, and when I restarted the prgoram now it just immediately closes after I get to the main window (after pressing start) and then the program just re-opens to the uninstall/start window.
[image]

Joromonni, 2020-12-08:
I did find another page of the tablet and it has listed the active area as 299 x 171mm (maybe its irrelevant idk). I also tried uninstalling and placing the config file into the folder before starting it, didn't change. As long as the DTK-1300 config was in the folder the program would just keep restarting itself after getting into the main window.

X9VoiD, 2020-12-08:
Did you extract the file? It should be a .json

Joromonni, 2020-12-08:
Okay now it boots up to the main window and doesnt close on start if the config is in there so maybe I had extracted incorrectly (sometimes I just drag files from the zip instead of properly extracting first). But now when I use detect tablet it closes.

X9VoiD, 2020-12-08:
If you want, you may join our [discord server](https://discord.gg/P3GYdUS) so we may assist you better.

Joromonni, 2020-12-08:
Joined.

InfinityGhost, 2020-12-10:
Any progress on this? Did we get a working configuration?

Joromonni, 2020-12-10:
I dont think so, I think we were waiting on some hid-thing to be added so we can troubleshoot/figure out more.

jamesbt365, 2022-06-18:
#2241

closed 2022-06-18 by jamesbt365