OpenTabletDriver/OpenTabletDriver #532: Left handed mode

Recilia, 2020-12-09:
This Driver Great!
but im having a hard time fully transferring to this driver due to no Left Handed mode.
Why need this you ask?
Well im a right handed but because of complex setup i have to put my CTL-472 on left handed so that the cable isnt all messed up
I know this feature will be added to the future but just bringing this up now so that i can actually try the new Driver with no problems.

Thank you. <3

AbstractQbit, 2020-12-09:
Already implemented in #528 , use action build from that branch to test it out (https://github.com/InfinityGhost/OpenTabletDriver/actions/runs/409491648), or set 180° rotation manually for now

Recilia, 2020-12-09:
thank you so much!

closed 2020-12-11 by InfinityGhost