OpenTabletDriver/OpenTabletDriver #535: Support Request: S620 Linux "ArgumentOutOfRange"

XSonicRU, 2020-12-10:
Hello there.
Idk bug or not, so issuing as support request
My S620 isn't being detected.
[image]
Have no idea how to fix - tried to generate new config from a connected s620 (gets detected as HUION tablet btw), and playing around with values.
Thanks in advance.  
btw im using arch

Mrcubix, 2020-12-10:
first, following the template would had been nice, 
second, 
```
sudo rmmod wacom; sudo rmmod hid_uclogic
sudo udevadm control --reload-rules && sudo udevadm trigger
```
then restart OTD

XSonicRU, 2020-12-10:
Sorry for not following the template, thought there was too much unneeded bloat there.
But, your advice has changed something apparently.
first two commands gave "module is not loaded"
After executing latter two I've restarted otd and nothing has changed.
I've tried to reinstall otd, just in case, and now I have a different error. Interesting.
[image]

XSonicRU, 2020-12-15:
So, I've done a little bit of tinkering there, have found some interesting details, although no solution.
First of all, the exception gets called by the lack of these values on one of HID devices
[image]
But the thing is, It actually has 2 instances of my tablet
[image]
And they are not identical. Bottom one has these values, and upper one doesn't. So, I thought that the issue might be that it just crashes on the upper one and doesn't even get to check the lower one. So, I've removed checks for these values in driver.
[image]
Hooray, no error, it detects successfully. But, driver doesn't seem to be working, no input. And neither it has any errors in its log. And it feels like I'm stuck here as I have no idea on how to debug the whole tablet-driver connectivity. I'm going to try tinkering with it a little more though, maybe will get lucky or anything.
Of course, any help would be welcome.

LavaDesu, 2020-12-15:
Are you using the AUR package or are you compiling and running it yourself?
This is most likely caused by misconfigured udev rules

XSonicRU, 2020-12-15:
Im using AUR.
But also compiling and running myself when debugging, right?
What's that even supposed to mean, lol

Udev rules? Never touched them myself though. Only did the stuff that the guy above advised, but it only reloads or something anyway.

XSonicRU, 2020-12-15:
Wait.
Just redone commands from above and 
IT WORKS.
Man I swear I was doing that before several times, for what godsaken reason did it only work now...
Why were they misconfigured it the first place though...
Anyway, thank you all.

closed 2020-12-15 by XSonicRU