OpenTabletDriver/OpenTabletDriver #537: driver opens and tries to connect to daemon then closes

2001tomcrowe, 2020-12-10:
this issue emerged today, it wasnt happening before.

tablet: gaomon s620
os: windows 10

2001tomcrowe, 2020-12-10:
this is the daemon log:
12/10/2020 06:31:06
Newtonsoft.Json.JsonReaderException
Unexpected character encountered while parsing value: {. Path 'OutputMode', line 2, position 17.
Newtonsoft.Json
   at Newtonsoft.Json.JsonTextReader.ReadStringValue(ReadType readType)
   at Newtonsoft.Json.JsonTextReader.ReadAsString()
   at Newtonsoft.Json.JsonReader.ReadForType(JsonContract contract, Boolean hasConverter)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.PopulateObject(Object newObject, JsonReader reader, JsonObjectContract contract, JsonProperty member, String id)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateObject(JsonReader reader, Type objectType, JsonContract contract, JsonProperty member, JsonContainerContract containerContract, JsonProperty containerMember, Object existingValue)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateValueInternal(JsonReader reader, Type objectType, JsonContract contract, JsonProperty member, JsonContainerContract containerContract, JsonProperty containerMember, Object existingValue)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.Deserialize(JsonReader reader, Type objectType, Boolean checkAdditionalContent)
   at Newtonsoft.Json.JsonSerializer.DeserializeInternal(JsonReader reader, Type objectType)
   at Newtonsoft.Json.JsonSerializer.Deserialize(JsonReader reader, Type objectType)
   at Newtonsoft.Json.JsonConvert.DeserializeObject(String value, Type type, JsonSerializerSettings settings)
   at Newtonsoft.Json.JsonConvert.DeserializeObject[T](String value, JsonSerializerSettings settings)
   at Newtonsoft.Json.JsonConvert.DeserializeObject[T](String value)
   at OpenTabletDriver.Settings.Deserialize(FileInfo file) in C:\Users\Infinity\OpenTabletDriver\OpenTabletDriver\Settings.cs:line 287
   at OpenTabletDriver.Daemon.DriverDaemon.LoadUserSettings() in C:\Users\Infinity\OpenTabletDriver\OpenTabletDriver.Daemon\DriverDaemon.cs:line 67
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__140_1(Object state)
   at System.Threading.QueueUserWorkItemCallbackDefaultContext.Execute()
   at System.Threading.ThreadPoolWorkQueue.Dispatch()
   at System.Threading._ThreadPoolWaitCallback.PerformWaitCallback()
ReadStringValue

InfinityGhost, 2020-12-10:
You switched to a development build and then reverted to an older version. In short, delete or move your settings at `%LOCALAPPDATA%\OpenTabletDriver\settings.json` and this issue will resolve itself.

InfinityGhost, 2020-12-12:
I'm closing this because I already know the root cause and it isn't a bug.

closed 2020-12-12 by InfinityGhost