OpenTabletDriver/OpenTabletDriver #541: Add support for Huion G10T

cantnot, 2020-12-10:
# Tablet Information

| Name            | Value |
| --------------- | ----- |
| Manufacturer    | HUION
| Product Name    | Huion G10T
| Width           | 374mm
| Height          | 177mm
| Pressure Levels | 8192
| LPI             | 5080

## Diagnostic Information

[w.txt](https://github.com/InfinityGhost/OpenTabletDriver/files/5676147/w.txt)

Mrcubix, 2020-12-10:
[Huion G10T.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5676157/Huion.G10T.zip)
put that config in configuration/Huion

cantnot, 2020-12-10:
HUION_T161_170531

Mrcubix, 2020-12-11:
[Huion G10T.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5676304/Huion.G10T.zip)

cantnot, 2020-12-11:
the cursor sticks to the top left corner of my monitor, it identifies my tablet as Huion GT-133 in the console, it does seem to be catching coordinates in the debugger, report id is 64 it doesnt change, it registers clicks from both buttons, but it doesnt work my cursor is always stuck in the top right corner

cantnot, 2020-12-11:
report ID seems to be always at 64, debug is registering clicks, pressure and position, when positioning my pen on the top it seems to get to the top on my screen before i get to the top with my tablet and i cant reach the bottom of my screen or the right side

Mrcubix, 2020-12-11:
you need to be slow else it just won't work,

PR #706 (2021-01-26): Add Huion G10T configuration
## Verification

#541 (Partially user verified, only tablet and sensor dimensions are not)

Tablet dimensions verification: [Huion official device page](https://www.huion.com/pen_tablet/InspiroyTouch/G10T.html)

`254mmx158.8mm` listed on the website is near the known `254mmx158.75mm`

## Issues

Closes #541
closed 2021-01-26 by InfinityGhost