OpenTabletDriver/OpenTabletDriver #546: Cursor jumping to the side when moving a pen

lisophoria, 2020-12-12:
Just installed this driver, configured an area and tried to play osu. The cursor is jumping somewhere like shown in picture below

[image]

This issue is systemwide.

Im runnin on Linux neon KDE that was built on ubuntu. My tablet is CTL-672. Any ideas?

InfinityGhost, 2020-12-12:
Could be a configuration issue, will have to cross check the 472 and 672 configs

X9VoiD, 2020-12-13:
This is a double cursor bug. Check wiki on how to fix that.

closed 2020-12-13 by lisophoria

Evernow, 2021-05-25:
If anyone else encounters this, in the wiki it is referenced as "ghost cursor" https://github.com/OpenTabletDriver/OpenTabletDriver/wiki/Linux-FAQ#i-have-a-ghost-cursor-constantly-alternating-cursor-position