OpenTabletDriver/OpenTabletDriver #550: Unable to locate tablet (HUION 430P) - Mac version 10.15.7

Alolann, 2020-12-13:
## Description
Unable to locate tablet 
## System Information:

| Name  Macbook Air 2020 | Value |
| MacOS Catalina 10.15.7 |
| Tablet     Huion 430p      |

InfinityGhost, 2020-12-13:
Export diagnostics from within the GUI (help -> export diagnostics) and send them here.

Alolann, 2020-12-14:
> Export diagnostics from within the GUI (help -> export diagnostics) and send them here.
Cant upload json here so i guess ill just copy and paste it
{
  "App Version": "OpenTabletDriver v0.4.2",
  "Operating System": {
    "Platform": 4,
    "ServicePack": "",
    "Version": "10.15.7",
    "VersionString": "Unix 10.15.7"
  },
  "Environment Variables": {
    "TMPDIR": "/var/folders/9p/p8607k3d5x94lxbslgmvj1p80000gn/T/",
    "XPC_FLAGS": "0x0",
    "SHELL": "/bin/zsh",
    "USER": "a46136",
    "__CF_USER_TEXT_ENCODING": "0x1F5:0x0:0x0",
    "SSH_AUTH_SOCK": "/private/tmp/com.apple.launchd.qe2LBevGm1/Listeners",
    "LOGNAME": "a46136",
    "XPC_SERVICE_NAME": "com.infinityghost.opentabletdriver.2248",
    "PATH": "/usr/bin:/bin:/usr/sbin:/sbin",
    "HOME": "/Users/a46136"
  },
  "HID Devices": [
    {
      "DevicePath": "IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/RP05@1C,4/IOPP/IOBC@0,1/IOBufferCopyController/AppleUSBVHCIBCE@80000000/AppleUSBVHCIPort@80400000/Headset@80400000/HID Relay@0/AppleUserUSBHostHIDDevice",
      "VendorID": 1452,
      "ProductID": 33027,
      "ReleaseNumberBcd": 533,
      "ReleaseNumber": "2.15",
      "ProductVersion": 533,
      "Manufacturer": "Apple",
      "ProductName": "Headset",
      "SerialNumber": "000000000000",
      "MaxInputReportLength": 2,
      "MaxOutputReportLength": 0,
      "MaxFeatureReportLength": 0
    },
    {
      "DevicePath": "IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC1@14/XHC1@14000000/HS01@14100000/Huion Tablet@14100000/IOUSBHostInterface@0/AppleUserUSBHostHIDDevice",
      "VendorID": 9580,
      "ProductID": 109,
      "ReleaseNumberBcd": 256,
      "ReleaseNumber": "1.0",
      "ProductVersion": 256,
      "Manufacturer": "HUION",
      "ProductName": "Huion Tablet",
      "SerialNumber": "",
      "MaxInputReportLength": 12,
      "MaxOutputReportLength": 0,
      "MaxFeatureReportLength": 0
    },
    {
      "DevicePath": "IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/RP05@1C,4/IOPP/IOBC@0,1/IOBufferCopyController/AppleUSBVHCIBCE@80000000/AppleUSBVHCIPort@80500000/Apple Internal Keyboard / Trackpad@80500000/HID Relay@0/AppleUserUSBHostHIDDevice",
      "VendorID": 1452,
      "ProductID": 640,
      "ReleaseNumberBcd": 2336,
      "ReleaseNumber": "9.20",
      "ProductVersion": 2336,
      "Manufacturer": "Apple Inc.",
      "ProductName": "Apple Internal Keyboard / Trackpad",
      "SerialNumber": "FM703950B2UN1QWAA+FML",
      "MaxInputReportLength": 5,
      "MaxOutputReportLength": 0,
      "MaxFeatureReportLength": 0
    },
    {
      "DevicePath": "IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/RP05@1C,4/IOPP/IOBC@0,1/IOBufferCopyController/AppleUSBVHCIBCE@80000000/AppleUSBVHCIPort@80600000/Touch Bar Backlight@80600000/HID Relay@0/AppleUserUSBHostHIDDevice",
      "VendorID": 1452,
      "ProductID": 33026,
      "ReleaseNumberBcd": 513,
[... 3258 lines cut ...]
      "Time": "2020-12-14T14:00:59.332683+07:00",
      "Group": "DeviceIOException",
      "Message": "Failed to open USB device: 0xE00002C5",
      "Level": 3
    },
    {
      "Time": "2020-12-14T14:00:59.332691+07:00",
      "Group": "Detect",
      "Message": "Searching for tablet 'Huion H610 Pro'",
      "Level": 1
    },
    {
      "Time": "2020-12-14T14:00:59.332699+07:00",
      "Group": "Detect",
      "Message": "No tablet found.",
      "Level": 1
    },
    {
      "Time": "2020-12-14T14:00:59.332707+07:00",
      "Group": "IPC",
      "Message": "Connected to a client.",
      "Level": 0
    }
  ]
}

InfinityGhost, 2020-12-15:
So its throwing code `0xE00002C5` (`kIOReturnExclusiveAccess`) when trying to open the IO stream, which means a driver already has exclusive access to the device. I don't know your current system configuration, but if you happen to have the official drivers or something installed, remove them. If not, I'm honestly not sure how to fix this.

InfinityGhost, 2020-12-15:
Linking this for development reference: https://developer.apple.com/library/archive/technotes/tn2315/_index.html

Alolann, 2020-12-15:
ah apparently i had 1 file from the huion driver that I havent deleted and it just fixed it. Thanks for the help!!! :)

closed 2020-12-15 by InfinityGhost