OpenTabletDriver/OpenTabletDriver #552: Driver not working after Sleep

C0D3-M4513R, 2020-12-13:
## Description

Open OpenTabletDriver
Set your computer to sleep
Wake the Computer
Watch yourself having to restart the OpenTabletDriver
## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System | Win 10 2004|
| Software Version | v0.4.2 |
| Tablet           | MTE-450 |

Mrcubix, 2020-12-13:
can't reproduce the issue, can even turn back on my pc with my tablet, OTD work directly

X9VoiD, 2020-12-13:
I just tested this too and it's not a bug in OTD.

X9VoiD, 2020-12-13:
G640s blinks its led indicator when the tablet itself detects a pen in range. That does not happen on the duration that the input is not working, and will suddenly work again when led blinks up again even without restarting OTD. This is a fault in the tablet itself not resetting fast enough after power resume.

C0D3-M4513R, 2020-12-13:
Can you supply a longer reset time in the config, or is there a plusing to prevent a tablet from going to sleep?

C0D3-M4513R, 2020-12-13:
the curious thing is, that even in the lock-screen the tablet is active(meaning, that it can be used, but without absolute positioning). The same goes for once you are logged in. Tablet buttons require more force when in that state. I should add, that a new detect tablets works as well to get the tablet back working. So maybe the tablet is being registered differently, or maybe OTD is losing the lock on the tablet, when sleeping, but is regaining it automatically for other tablets?

InfinityGhost, 2020-12-14:
> Can you supply a longer reset time in the config, or is there a plusing to prevent a tablet from going to sleep?

Fwiw, OpenTabletDriver doesn't do any of the "keep alive" stuff, the kernel USB stack does. This seems more like an operating system issue more than an OTD issue.

C0D3-M4513R, 2020-12-14:
Maybe a solution to this is to just re-detect tablets on restarting from sleep?

InfinityGhost, 2021-04-06:
Not worth fixing, its really out of scope of the driver to do this.

closed 2021-04-06 by InfinityGhost