OpenTabletDriver/OpenTabletDriver #553: Tablet support on admin prompt

C0D3-M4513R, 2020-12-13:
## Description

When OpenTabletDriver is active, the tablet will stop working during a admin permission promt.
More specifically, the mouse will just be frozen in place, when using the tablet as an input source
I would like for that to change, and have OpenTabletDriver work during a admin promt.

Can't record admin elevation promt

AbstractQbit, 2020-12-13:
Duplicate of #347, #469 and #539.

X9VoiD, 2020-12-13:
If you really need input on an elevated process then use kernel-level mode plugins like VMultiMode from VoiDPlugin repo

closed 2020-12-13 by InfinityGhost

C0D3-M4513R, 2020-12-13:
VmMultiMode may be a good plugin, but it just breaks tablet detection. When I use it, my tablet is just instantly unrecognised

X9VoiD, 2020-12-13:
It *never* affects tablet detection when you've done everything correctly. Make sure you've followed the wiki properly to install vmulti dependency.

C0D3-M4513R, 2020-12-14:
```
Adding driver package:  vmulti.inf
Driver package added successfully. (Already exists in the system)
Published Name:         oem31.inf
Driver package is up-to-date on device: ROOT\HIDCLASS\0000
```
Don't see any issue there...