OpenTabletDriver/OpenTabletDriver #564: Support for H1060P?

czarchasm00, 2020-12-17:
# Support Request
I see that the Huion New 1060 Plus is listed as being supported in your Supported Tablets list. However, the Huion H1060P is not. (AFAIK) The notable difference between the two is that the H1060P does not have a battery in its pen that you have to charge or replace, and it also has 8192 pressure levels. The H1060P is commonly known as the "upgraded version" to the New 1060 Plus (AFAIK), so I am unaware if the drivers are different.
My main question is this: Are the H1060P and New 1060 Plus using the same drivers and this is an issue on my end, or will the H1060P require its own driver support.

Notably, the TabletDriver project from hawku is able to detect that it is a Huion tablet, and my pen's position (using debugger), but it does not know which one it is or actually put inputs through, so I'm thinking it does indeed use different drivers.

## Diagnostic Information
Help -> Export Diagnostics: https://pastebin.com/kBZ8UsLt

X9VoiD, 2020-12-17:
OTD -> Tablet -> Device string reader
```
Vendor ID: 9580
Product ID: 109
String Index: 201
```
Hit Send Request, and then paste result here

czarchasm00, 2020-12-17:
HUION_T167_190325

X9VoiD, 2020-12-17:
Make sure you don't have any tablet drivers anymore. If you do, please uninstall them first including WinUSB.

Download latest [dev build](https://github.com/InfinityGhost/OpenTabletDriver/actions/runs/426981263). Extract it to any folder of choice as long as it's not an admin-requiring folder.

Extract [H1060P.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5706866/H1060P.zip) as `Configurations\Huion\H1060P.json` and then launch `OpenTabletDriver.UX.Wpf.exe`

czarchasm00, 2020-12-17:
I have already removed the "Graphics Tablet Device" from the Device Manager, as well as ticking the box to uninstalling it. I've uninstalled the Huion official drivers through uninstalling the official driver program. I'm not entirely sure what WinUSB is; is that another driver? I don't seem to have anything named "WinUSB" in my Device Manager.

Regardless of that, your instructions worked perfectly. After installing the new .NET 5.0 Runtime, my tablet is recognized properly with the new dev build and I am able to use it without any issues (so far). Thank you so much.

closed 2020-12-17 by czarchasm00

X9VoiD, 2020-12-17:
For verification processes, go to Tablet -> Tablet Debugger, hover your pen around the edges of the tablet and try to get the highest possible value for Position

czarchasm00, 2020-12-17:
I got both of those exact values. Apologies for closing this issue early.
[image]
Let me know if there is any additional info you would like me to provide.

X9VoiD, 2020-12-17:
No problems, we're good, H1060P support will come on next official release :bread:

For now you may use the dev build that you currently have, and if you want to update your dev build, go [here](https://github.com/InfinityGhost/OpenTabletDriver/actions), find latest `master`, and then place the config I sent you.