OpenTabletDriver/OpenTabletDriver #566: Support Request: Tablet not detected [Star G640] [Arch Linux]

straightcurve, 2020-12-17:
# Support Request
I followed the Arch Linux instructions, no problems encountered during the installation, however the tablet is not recognized.

## What I've already tried
I was using the wacom driver for the tablet before finding out about OTD, I tried removing the from /etc/X11/xorg.conf.d/52-tablet.conf from there.
I tried restarting the machine.
Executing `systemctl status --user opentabletdriver.service` gives me this:
[image]
I don't know where to find the logs haha

## Diagnostic Information
I tried to export diagnostics, but it's coming out as a 0 bytes file.
`uname -a
Linux 5.9.11-arch2-1 #1 SMP PREEMPT Sat, 28 Nov 2020 02:07:22 +0000 x86_64 GNU/Linux`

InfinityGhost, 2020-12-17:
Run `journalctl --user-unit opentabletdriver.service` and try to get that full exception in text.

straightcurve, 2020-12-17:
[image]

X9VoiD, 2020-12-18:
Replug tablet or retrigger udev through `sudo udevadm trigger` and then restart OTD daemon and UX

straightcurve, 2020-12-18:
nothing changed, also this is what the console says:
[image]

X9VoiD, 2020-12-18:
From: https://github.com/InfinityGhost/OpenTabletDriver/wiki/Linux-FAQ
```
echo "blacklist hid_uclogic" | sudo tee -a /etc/modprobe.d/blacklist.conf
sudo rmmod hid_uclogic
```

straightcurve, 2020-12-18:
I have not seen the FAQ page, thank you, it works now.

closed 2020-12-18 by straightcurve