OpenTabletDriver/OpenTabletDriver #568: Driver does not work on fullscreen applications started with administrator privileges.

Arthur095, 2020-12-18:
## Description

Title says all. I don't know if it's intended or not so I post it as bug.
## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System Windows 10 1909 |
| Software Version Last version as 2020/12/18 |
| Tablet   CTL-4100-S        |

X9VoiD, 2020-12-18:
Duplicate of #347, #469, #539 and #553

X9VoiD, 2020-12-18:
Please read the other duplicate issues for more information as to why OTD will never support injecting input to elevated processes for security reasons. This is *intentional* and can't ever be "fixed". If you really need to inject input to elevated processes then install a kernel-level driver like [VMulti](https://github.com/X9VoiD/VoiDPlugins/wiki/VMultiMode).

This issue should be closed.

closed 2020-12-18 by Arthur095