OpenTabletDriver/OpenTabletDriver #569: Support Request: Tablet is not being detected

theokguy35, 2020-12-19:
# Support Request

My tablet is not being detected, A Supported GAOMON S620 I use Pop_OS and it isn't being detected..

## What I've already tried
I have tried opening it in the terminal, opening it on sudo, I'm not a Linux expert, I've been using it for some time now, Replugging, Changing cables.

## Diagnostic Information
[diagnostics.txt](https://github.com/InfinityGhost/OpenTabletDriver/files/5718925/diagnostics.txt)

Thanks!

theokguy35, 2020-12-19:
The dependencies are all installed by the way,

InfinityGhost, 2020-12-19:
Looks like hid-uclogic might be active. Try [this](https://github.com/InfinityGhost/OpenTabletDriver/wiki/Linux-FAQ#argumentoutofrangeexception-value-0-15), and please report back whether it worked or not.

theokguy35, 2020-12-19:
Had to do it twice but it worked instantly!
Thank you very much dude!
[itworked.txt](https://github.com/InfinityGhost/OpenTabletDriver/files/5718947/itworked.txt)

closed 2020-12-19 by theokguy35