OpenTabletDriver/OpenTabletDriver #570: Area calculator suggestion.

brotherhooder, 2020-12-19:
### Area calculator
Just a little suggestion.
### Suggestion:
You know original driver users? Yep, It's undeniable that there is a lot of them, And! With time, Most of them will want to get a new tablet, Or simply get better drivers, And here it enters the calculator, They have their areas and they want to use their areas on new drivers, Or on new tablets, If they wanted to use the driver, They would be greeted with a easy to configure area, Let's say that user came with a Huion tablet, BOOM!, Huion area values, This would attract those people to custom tablet drivers, And help develop the drivers by reporting bugs, More suggestions and on, Users feel demotivated when they see they cant use their original area, Also being maybe the first tablet drivers to support multiple area values(Hawku only supports Wacom which isn't exactly a very cost friendly company)

### Conclusion

- Community growth

- More bug reports & fixes

- User friendly enviroment

- Active community.

Thanks for reading!

InfinityGhost, 2020-12-19:
We plan on doing that! It'll automatically use detected tablet's vendor to determine which converter to use to simplify the end user experience.

brotherhooder, 2020-12-19:
Sounds awesome!

PR #579 (2020-12-30): Add Area Converter window
Closes #570
closed 2020-12-30 by InfinityGhost