OpenTabletDriver/OpenTabletDriver #572: Area automatically changing upon restart of driver.

Myekel, 2020-12-19:
## Description

Setting my area to
Width: 67
Height: 42
Xpos: 114.54
Ypos: 70.83
Rotation: 180
I get this weird thing where it automatically changes to itself to (upon restarting)
Width: 67
Height: 42
Xpos/offset: 118.5
Ypos/offset: 74
Rotation: 180
Could this get looked at? is withholding me from actually using the drivers since its really annoying to continuesly fix my area.

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System | Win 10 2004
| Software Version | 0.4.2
| Tablet           | CTH-480

Mrcubix, 2020-12-19:
make your area, save it, upload your settings.json ( found in %localappdata%/opentabletdriver/ )
restart OTD, save, and upload settings.json for the second time

Myekel, 2020-12-19:
Okay so you want me to upload them here?

Myekel, 2020-12-19:
[settings1.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5719575/settings1.zip)
[settings2.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5719576/settings2.zip)

Myekel, 2020-12-19:
Oh btw, When changing area position values to the values above they get changed again when putting rotation to 180.

Mrcubix, 2020-12-19:
this seem to be in reality, a UX bug, the values are the same, as well ass probably for the daemon too

Mrcubix, 2020-12-19:
you're using 180 rotation cause you're left handed?

Mrcubix, 2020-12-19:
https://cdn.discordapp.com/attachments/615611007951306863/786267880207417374/Lefty_Flip_Mode.dll

put this in your plugin folder

Myekel, 2020-12-19:
I use 180 cuz there is a scratch on the topleft of my tablet so i flipped my tablet upside down to still play with topleft.

Myekel, 2020-12-19:
When using the plugin it still gets set to x: 118.5 y: 74

Myekel, 2020-12-19:
nvm
I now have to put my area to the topleft of my tablet of course.

AbstractQbit, 2020-12-19:
@Myekel could you try reproducing your issue on the latest dev build? Because I don't encounter such behaviour there, it saves exactly what's in the field.
https://github.com/InfinityGhost/OpenTabletDriver/actions/runs/431969720

Myekel, 2020-12-19:
I will try it

Myekel, 2020-12-19:
Allright! Seems to be working flawlessly! thank you for help!

closed 2020-12-20 by InfinityGhost