OpenTabletDriver/OpenTabletDriver #576: Will this work on macOS Big Sur?

jklmao, 2020-12-20:
## Description

Hi, I recently updated my mac and now running Big Sur. I was just wondering if future support for Big Sur will come available because everytime I try running OpenTabletDriver, it refuses to open.
## System Information:

| Name             | Value |
| Chris | ----- |
| macOS Big Sur |
| Latest |
| Veikk s640|

InfinityGhost, 2020-12-21:
There's a couple known bugs in Big Sur related to a dependency of OpenTabletDriver. It *may* work by v0.5.0, but no guarantees because we don't have any way to test it right now.

JWOK0, 2021-01-20:
I'm having the same issue with Big Sur, installed the .NET 5 Runtime but I'm still getting the same error message while trying to open it 
[image]

Hopefully you can get it working, but do take your time

InfinityGhost, 2021-01-20:
> I'm having the same issue with Big Sur, installed the .NET 5 Runtime but I'm still getting the same error message while trying to open it
> [image]
> 
> Hopefully you can get it working, but do take your time

Will fix soon, something bugged out with my packaging script.

InfinityGhost, 2021-01-20:
So I fixed the packaging, but we have a major breaking bug right now on MacOS, #690. It will have a fix with v0.5.1

InfinityGhost, 2021-05-25:
Issue appears to be dead, haven't had many reports from big sur users either. Assuming resolved.

closed 2021-05-25 by InfinityGhost